juicejuice/homeassistant_redback #1: SH4600 Inverter

djgoding, 2023-03-22:
Hi there,
From your notes
If you have a different model I would be interested to adapt the code to suit, but would need some data packets from your system

I have a SH4600 Redback inverter and keen to get it integrated.
What information do you require?

juicejuice, 2023-03-22:
Sounds good! Firstly you need API access, which you can request from Redback support team. Then you will have access to the Swagger API site: https://api.redbacktech.com/docs/index.html. From there you need to retrieve two EnergyData samples, one for Static and one for Dynamic data. Please sanitise those samples (they may contain some personal info) then send them over. That should be enough for me to check against.

djgoding, 2023-03-22:
Static and Dynamic Data as requested
[Static and Dynamic Data Redback SH4600.zip](https://github.com/juicejuice/homeassistant_redback/files/11044753/Static.and.Dynamic.Data.Redback.SH4600.zip)

juicejuice, 2023-03-29:
Thanks @djgoding for the sleuthing here. I've now pushed an update to the integration code which will detect whether the inverter has any connected batteries. If there are no batteries, it will not create the battery entities. It seems to work here for me but my system has batteries, so please try the updated integration out when you get chance and let me know?

LachyGoshi, 2023-03-29:
I can test this out over the weekend also and see how it runs on my si5000 inverter.

LachyGoshi, 2023-04-01:
Just re installed this integration via hacs and as expected, it picked up my SI5000 inverter (and not my SH5000 Hybrid) and looks like it has not added any battery entities, after about a minute, all the appropriate inverter entities did populate OK!

so this does look like you no longer have to disable the battery entities when you don't have batts connected for all the other data to show anymore!

Nice one Juice!

juicejuice, 2023-04-02:
Thanks for the confirmation, I'll mark this issue as closed then :)

closed 2023-04-02 by juicejuice