juicejuice/homeassistant_redback #2: Multiple Inverters

LachyGoshi, 2023-03-22:
Hi, thanks for this great integration.

I have two redback inverters.
An SI5000 standard single phase grid tied inverter
An SH5000 hybrid single phase with 21kwh battery

I received my ID and Key from redback, and following their API user guide I believe both inverters are available via that one ID/key.

However the home assistant integration only seems to detect the first grid tied inverter.... But not really present any useable info as all the device entities values are unknown.... The device si5000 model and firmware version (unconfirmed if the firmware is actually correct) is all that appears to successfully populate. 

Can I force the integration to look for a specific inverter serial number or to autoscan for multiple devices/serial numbers?

I can also share info on the si5000 of you want to analyse to try and get compatible (I mainly just want the hybrid to work)

Thanks you.

djgoding, 2023-03-24:
Hi Lachy,
I was seeing a similar view on my SH4600 (No batteries)
I went into the entities under the Redback integration and disabled all of the battery options, it then presented all of the other information, Grid, Solar etc

juicejuice, 2023-03-24:
Thanks guys, I will check the latest data packets out, been busy this week with work though. @LachyGoshi can you please send some data samples? I can then see how to support multiple inverters. From Redback's Swagger API site (https://api.redbacktech.com/docs/index.html) you need to retrieve two EnergyData samples, one for Static and one for Dynamic data. Please sanitise those samples (they may contain some personal info) then send them over. That should be enough for me to check against.

LachyGoshi, 2023-03-26:
Thanks DJ, making all the battery related entities did start showing the grid tied inverter settings like grid voltage etc ok.

Hi Juice, find API responses attached. It seems I have two sites.... Which essentially is each inverter under each site... It would appear that if I could either specify the serial number or the site ID into the integration then that could potentially force a individual device in home assistant per inverter, I could then enable/disable the appropriate entities per device....

Thanks for your help, I really appreciate it. 
[Redback api lachygoshi filtered.zip](https://github.com/juicejuice/homeassistant_redback/files/11071242/Redback.api.lachygoshi.filtered.zip)

juicejuice, 2023-03-29:
Haha, all good! I'm just pleased this is of value to others. I was always going to use it myself but now it seems more worthwhile :)

LachyGoshi, 2023-04-01:
Took me longer than expecte to get this done, but here are the sites showing each inverter/// one per site.
[response_1680383320406.txt](https://github.com/juicejuice/homeassistant_redback/files/11130609/response_1680383320406.txt)

LachyGoshi, 2023-04-01:
as a side note, I just tried the "Private" connection method to see if I could force a connection to the SH5000 hybrid, but just keeps giving authentication error - not surprised as per your notes in the setup info, but I thought it would be worth a try.

It does look like that private method no longer works.

juicejuice, 2023-04-02:
Yeah, private needs an API token but it sounds like I should remove that access method. I only left it there for testing, really. Seeing as Redback are happy to supply API credentials to any customer then there is no need for that "backdoor" API any more.

juicejuice, 2023-04-02:
> Took me longer than expecte to get this done, but here are the sites showing each inverter/// one per site. [response_1680383320406.txt](https://github.com/juicejuice/homeassistant_redback/files/11130609/response_1680383320406.txt)

Thanks, that's great! The API response looks exactly like I thought it would. Give me a little time due to Easter etc. but I should be able to add a pretty simple mechanism for selecting which site to add. It won't be difficult, I'll just add another field during setup where you can indicate which site to add.

juicejuice, 2023-04-06:
I've added the support for multiple inverters now, @LachyGoshi. It works here for my single inverter, but can you please test and let me know? I've updated the README docs to describe how it should work :)

LachyGoshi, 2023-04-07:
Thanks @juicejuice I will test it out over the Easter weekend and shall report back. Have a good long weekend.

djgoding, 2023-04-07:
Hi there juicejuice, I am no longer able to select any of the live redback entities, ie. Redback Grid Export, when configuring the energy dashboard. I can only select the totals, ie. Redback Grid Export Total, from the dropdown. These are all showing in the integrations\redback\entities and also in the overview page. I've done a full reinstall of  the integration however this did not resolve the issue.

LachyGoshi, 2023-04-10:
Hi @juicejuice, I've just had a chance to test the latest version.

I get best results when adding the "second" site to HA first. This is my Hybrid SH5000 with the batteries. This populates all the entities as expected.

Then, when I go to add in the "first" site which is just my standard grid tied SI5000 inverter (No batteries), unfortunately no entities populate. This seems to be due to the integration creating the entities based on the Redback API Client ID, rather than by Site ID or Serial number.

IF I delete everything and start from scratch, and add in the "first" site first, this populates all the SI5000 entities, then I go and add in the "second" site, this actually then adds all the battery related entities into the "first" SI5000 device page, and jumping between both devices makes HA look very confused. Hard to explain but let me know if you need me to record screen, but I think if you change that [entities](url) get a unique ID based on Site ID or Device Serial Number, then that would probably resolve this anyway.

Screenshot 1 and 2: After adding in the "second" site first which is the SH5000, All looks OK with 1 device and 15 entities:
[image]
[image]

Screenshot 3: After adding in the next site (the "first" site) which is the SI5000, No Devices or Entities list:
[image]

Screenshot 4: Log entry showing that replicated entities based on the Redback API Client ID cannot be created:
[image]

juicejuice, 2023-04-10:
Yes you're right, it will need the IDs adjusting. I think it should also work if you give each device a unique "friendly" name, again hard for me to test here. I'll look into the uniqueness for the IDs though, that should be all that's needed to fix this.

juicejuice, 2023-04-10:
@LachyGoshi thanks again for being the guinea pig and for your detailed issue report. Sorry my first attempt at a fix went down in flames :)
I've now pushed an update that should resolve the issue. Each entity is now assigned a unique ID based on the Site ID and not on the Client ID. That approach makes a lot more sense and was already on my TODO list. Please test again and do let me know if you find any other bugs!

juicejuice, 2023-04-10:
> Hi there juicejuice, I am no longer able to select any of the live redback entities, ie. Redback Grid Export, when configuring the energy dashboard. I can only select the totals, ie. Redback Grid [...]

Hi @djgoding what you describe is actually the design of HA Energy dashboard and is in the description when you add an entity "Pick a sensor which measures XXXXX in either of Wh, MJ, MWh, kWh, GJ." The sensors have to be energy totals, not an instantaneous power reading. That's why I've provided energy total sensors for all the necessary data points for the energy dashboard.

LachyGoshi, 2023-04-10:
> Yes you're right, it will need the IDs adjusting. I think it should also work if you give each device a unique "friendly" name, again hard for me to test here. I'll look into the uniqueness for the IDs though, that should be all that's needed to fix this.

I have tried different friendly names, same result.

LachyGoshi, 2023-04-10:
> @LachyGoshi thanks again for being the guinea pig and for your detailed issue report. Sorry my first attempt at a fix went down in flames :) I've now pushed an update that should resolve the issue. [...]

Amazing mate, I shall test when able and report back!

LachyGoshi, 2023-04-11:
Amazing.... On first very quick look it all seems to be detecting both inverters properly but I'll do some full analysis this weekend to make sure none of the fields are doubled up and are tracked to the correct inverter. I shall report back if all ok to close the ticket!

[image]

[image]

[image]

[image]

[image]

LachyGoshi, 2023-04-11:
Did find this in the logs even though both inverters reporting grid voltage ok:

Logger: homeassistant.components.sensor
Source: custom_components/redback/entity.py:19
Integration: Sensor (documentation, issues)
First occurred: 19:13:57 (1 occurrences)
Last logged: 19:13:57

Error while setting up redback platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 293, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/redback/sensor.py", line 180, in async_setup_entry
    RedbackVoltageSensor(
  File "/config/custom_components/redback/entity.py", line 19, in __init__
    site_id = coordinator.config_entry.data["site_id"]
KeyError: 'site_id'

juicejuice, 2023-04-11:
Thanks - I'll look into it. My testing here with a clean HA instance didn't show that up so it might be to do with an HA environment that was running the old version of my code. It has probably created a "config entry" without the site_id in it, whereas that field is now required by the new version of the code. I'll see what can be done!

juicejuice, 2023-04-15:
@LachyGoshi I've pushed another update now. I discovered that HA provides a mechanism for migrating the Redback device "config entry" as the plugin is updated. So even though Redback entities are now based on the site_id (allowing multiple inverters), any existing Redback devices that were setup with my old code will now continue to work as expected. That should get rid of the errors you are seeing in the log! Let me know how you get along?

LachyGoshi, 2023-04-17:
Thanks Juice. I'm away this week so will test it out when I'm back home in a week

LachyGoshi, 2023-04-28:
Hi Juice, Thanks mate, I have just removed the old devices and integration and then reinstalled it via hacs and re added both inverters. Both populated as expected (SH5000 with battery entities, and SI5000 without battery) and no errors found in logs!

I was experiencing days when the days consumption graphs in the home assistant Energy dashboard would only show one or two ridiculously large columns of consumption at midday for each day, other days remained looking ok with the more expected bell curve shape. I will continue to monitor this new version (and I have also just updated my HA and OS, and will let you know if any other issues arise!

Thanks again now all we need to hope for is a more open system from Redback and some actual responsiveness from their support :)

juicejuice, 2023-05-11:
Thanks @LachyGoshi for the feedback! I'll close this issue as we now seem to have multiple inverters working nicely.

closed 2023-05-11 by juicejuice