juicejuice/homeassistant_redback #3: Charge HQ Integration

djgoding, 2023-03-22:
Hi there,
I see that ChargeHQ list your code as a push API integration
https://chargehq.net/kb/push-api
Do you have some further instructions on how this is setup using HomeAssistant to publish the data to them?

juicejuice, 2023-03-24:
Oh I didn't realise I was linked from ChargeHQ, thanks! I briefly discussed with Andrew from ChargeHQ and he was going to talk with Redback about a proper integration. I'm not sure how much progress he made.
It looks like the ChargeHQ push API lets you feed inverter data into their system - cool! I ended up implementing my own equivalent of ChargeHQ in Home Assistant. It wasn't that hard, I can post up a Youtube video if you like?
Otherwise it looks like you would have to implement some HA scripting to send the Redback entity data over to ChargeHQ. I haven't tried that so don't have any sample code or instructions, sorry.

djgoding, 2023-03-25:
Yep I'm keen to see what you have implemented

djgoding, 2023-04-18:
Hi there juicejuice,
I have been doing some further testing with an aim to export the redback solar information to Charge HQ and I was wondering if I am heading down the correct path, please note I am not a programmer, most of this is trial and error.
My thoughts are that I need to create a sensor that pushes the redback inverter values in a JSON payload to ChargeHQ. I guess I will then need to run a script to activate the POST every minute or so?
I have added the following to my configuration.yaml and restarted, however the SENSOR does not get created after I restart and I get the following recurring error in the log, Setup of sensor platform rest is taking over 10 seconds.

sensor:
  - platform: rest
    name: My Redback JSON Data
    resource: https://api.chargehq.net/api/public/push-solar-data
    method: POST
    headers:
      Content-Type: application/json
    payload: >
      {
        "apiKey": "abc12345678901234567890",
        "siteMeters": {
          "consumption_kw": {{ states('sensor.redback_sh4600_inverter_load_total') | float }},
          "net_import_kw": {{ states('sensor.redback_sh4600_inverter_grid_import') | float }},
          "production_kw": {{ states('sensor.redback_sh4600_inverter_solar_generation') | float }},
          "exported_kwh": {{ states('sensor.redback_sh4600_inverter_grid_export_total') | float }},
          "imported_kwh": {{ states('sensor.redback_sh4600_inverter_grid_import_total') | float }}
        }
      }
    json_attributes:
      - siteMeters

djgoding, 2023-04-18:
This is how the script is actually formatted
[image]

djgoding, 2023-04-18:
Here are the Redback entity IDs that I am referencing in the payload states
[image]

juicejuice, 2023-05-11:
Hi @djgoding, sorry but I'm going to close this issue as I've written my own solution for this in HA. I did find this thread though in case it helps? https://community.home-assistant.io/t/pvouput-uploader/46761

closed 2023-05-11 by juicejuice