juicejuice/homeassistant_redback #5: Charge / Discharge / Export / Import Settings

LachyGoshi, 2023-04-10:
Another request / guidance....

I have Successfully be been able to set a schedule via the API to set my system to charge the battery for the next 15 minutes and then return to Auto mode.

This has the potential to be useful to charge the battery in offpeak time especially during the winter months or during prolonged rainy days/weeks.

Is there any way via this integration to call a service via the API to great a schedule based on the below? Would be great to come up with a way to automate and future predict best charge time etc based on solar forecast and user input such as holiday mode etc to maximise export when you know you wont need the power yourself. Has anyone tried this before and can point me in the right direction?

[Schedule] POST [​/Api​/v2​/Schedule​/Create](https://api.redbacktech.com/docs/index.html#/Schedule/Schedule_CreateSchedule)
Creates a new schedule for a site

{
  "SiteId": "S2942XXXXXXX",
  "UserNotes": "Charge",
  "StartTimeUtc": "2023-04-10T04:30:00.000Z",
  "Duration": "00:15:00",
  "DesiredMode": {
    "InverterMode": "ChargeBattery",
    "ArgumentInWatts": 4000
  }
}

In my test scenario today, in the My Redback dashboard, I had the system "Settings -> Inverter -> Set -> Auto" which is usually my default mode. At 2:25pm (Melbourne Time) I then executed the above POST command via the API website a few minutes before this was due to start which was 2:30pm (just to give the system a few minutes to make sure everything synced ok - I think 1 minute lead time appears adequate). As the name suggests, start time is UTC time so add 10 hours for my local time. Refreshing the My Redback Dashboard showed that the "Settings -> Inverter -> Schedule" was now active and my new schedule was listed below.

Sure enough, when 2:30pm rolled around, the system flicked into Charge mode as expected as seen in "Analysis -> System Details -> Inverter Mode - Discharge".

After 15 minutes, although the My Redback Dashboard was still showing "Settings -> Inverter -> Schedule", the system itself had returned to Auto, as was evident by "Analysis -> System Details -> Inverter Mode - Auto".

I think there must be a way to layer schedules, for example, have the system set to Auto like I did, specify Schedule1 to charge from 10am for 6 hours, but you could specify Schedule2 to run from 12:00 for 2 hours to a discharge (I know not very real world but just trying to make a simple example here) where I think the system would be in auto, switch to charge at 10am, then discharge at 12pm, back to charge at 2pm and then return to auto at 4pm. I will test this out next weekend.

Also worth considering would be how to configure for recuring schedules within home assistance, as these start times are date specific, so the system would have to generate it each day, but I could see myself having a dashboard showing what mode its currently in, and then every 10 minutes or so, an automation runs to consider various factors and change the mode if required. The API does return an ID for the new schedule which would be useful to delete and re apply a new schedule so would need home assistance to track this ID somehow.

Response body
Download
{
  "Data": {
    "ScheduleId": "171c356e-7c41-4797-b40f-9d5e8520fea5"
  }
}

I have just received a TCP <-> CAN Bus which I am going to try and inset between the Pylontech batteries and the Ouja board in the SH5000 which may or may not give control of the system following some of these similar projects:
https://community.home-assistant.io/t/universal-solar-inverter-over-modbus-rs485-tcp-custom-component-growatt-sofar-solax-solis/140143
and
https://community.home-assistant.io/t/getting-data-from-solis-inverter/302189/44

If this works then it might mean we wont need to go via the API and schedules to change mode. But this project will take me a while.

juicejuice, 2023-04-10:
Great idea! I'll look into it when I get time. I believe the Redback inverters are Goodwe models, so in theory you should be able to use something that works with that brand? You definitely need a device that sits alongside the Ouija board because it hides all network traffic and only sends secure (encrypted) MQTT outbound.

LachyGoshi, 2023-04-10:
I less we could stack the SSH password to the Ouija board.... I've asked redback support but they won't supply it to me.

juicejuice, 2023-04-16:
Just leaving a few notes here. In the Redback portal you can set the mode indefinitely (e.g. no end time) and also you can set a schedule to run daily. It's not obvious how to achieve that via the API.

closed 2023-05-11 by juicejuice