juicejuice/homeassistant_redback #6: Battery Issues?

Lastaerie, 2023-04-11:
Hi, Firstly thanks for the effort on this.  I've been asking RBT about this for aaaages and tried myself a year or so ago but didn't get very far.
My issue is this.  is there a way to return the information relating to batterys?  i have 2 battery packs in my system and noticed today (while setting this up) that one of the packs has stopped working according the to the console.  As there is no notification method in the Redback portal it would be cool to have something set up via HA.
James

juicejuice, 2023-04-15:
I like the idea, here is what the Redback API provides in terms of battery details:
- "MinSoC0to1" = minimum allowable charge for battery (e.g. 20%)
- "MinOffgridSoC0to1" = minimum allowable battery charge when the grid is down
- "BatteryMaxChargePowerkW" = maximum battery charge rate
- "BatteryMaxDischargePowerkW"= maximum battery discharge rate
- "BatteryCapacitykWh" = total installed battery capacity
- "UsableBatteryCapacitykWh" = total available battery capacity
- "BatteryModels" = list of installed batteries
- "BatterySoCInstantaneous0to1" = current battery charge (SoC = state-of-charge)
- "BatteryPowerNegativeIsChargingkW" = current battery charge/discharge rate

Which of those would be suitable for your needs? I haven't had a battery stop working so I don't know what that would look like from the API perspective? What does it look like on the Redback portal? Final thought, you could test out the API methods and report back? https://api.redbacktech.com/docs/index.html

Lastaerie, 2023-04-15:
The only notification that i got was in the redback web portal it tells you what the inverter type, panels capacity and battery capacity. i noticed it was only saying 3.6Kwh instead of 7.2.  I'll have a play around with the API methods and see which one looks right "Battery Capacity" looks like it might be the right option. but will confirm.  the is only shown on the web portal and not the mobile app. Thanks  J.

juicejuice, 2023-04-16:
@Lastaerie sounds good. In that case I think it should be easy to pass through the relevant data points. For example, if BatteryCapacitykWh and UsableBatteryCapacitykWh were available as HA entities then you could write some automation that checks those entity values are within expected parameters?

Lastaerie, 2023-04-16:
Hi,  
is it possible to pass through the following 3:
1. "BatteryCapacitykWh" = total installed battery capacity
2. "UsableBatteryCapacitykWh" = total available battery capacity
3. "BatteryModels" = list of installed batteries

1 & 2 would enable a reconciliation of battery performance and 
1 & 3 would help to a confirm the availability of batteries but also assist in the diagnosis of a fault assuming that 3 is a dynamic & reported list rather than a installer set field at the point of install.
 thanks -James

juicejuice, 2023-05-11:
Sorry for the delay @Lastaerie, I'm going to add a few other entities (perhaps on the weekend) so I'll include these in the mix.

Lastaerie, 2023-05-11:
That's all good and thankyou. I'm just super stoked you are working on this integration at all.  It's adding so much value to my Solar setup.  Redback should be sponsoring you.

juicejuice, 2023-05-13:
Hi @Lastaerie, I've looked into this today but realised what you're asking for might not actually do what you want. Those three fields you identified are all from the StaticData API. The StaticData doesn't change, it's configured by the installer when your system is installed. Can you check with the API and see if "UsableBatteryCapacitykWh" changes for your system? Mine seems to stay the same, but maybe I'm missing something.

Lastaerie, 2023-05-13:
hi @juicejuice , will do.  I assume there must be some method reporting back as the web portal displays battery capacity which is how i am currently seeing the occasion occurrence of my battery capacity flipping between 7.1 & 3.6 Kwh.

juicejuice, 2023-05-13:
Yes that did occur to me too, but then again I have noticed the portal+app use a different "private" API as compared to the public one this integration is using.
Anyway, please do report back if you can find a way via the regular public API to retrieve the data you need?

Lastaerie, 2023-05-13:
Just thinking that if this is not the case I will raise a request to see if it can be added to the public API.  Seems odd that they wouldn't provide it by default.

juicejuice, 2023-05-13:
Yeah, makes sense to have detailed real-time battery information beyond the basic charge/discharge. Anyway, let me know how you get along :)

Lastaerie, 2023-05-13:
Hi @juicejuice looking at postman today it looks like this value is changes:

delivered from the Static data set

7am
"BatteryCapacitykWh": 7.104,
"UsableBatteryCapacitykWh": 6.394,

11am
"BatteryCapacitykWh": 3.552,
"UsableBatteryCapacitykWh": 3.197,

Kind regards

James

juicejuice, 2023-05-14:
Good research! I'll have a look at pulling that data through.

juicejuice, 2023-05-15:
@Lastaerie please check out my latest commit, it adds two new sensors for the values you found were changing. Does this work for your needs?

Lastaerie, 2023-05-15:
@juicejuice I'll update now and then add the fields to check. Many thanks for this will update shortly.

Lastaerie, 2023-05-16:
Hi @juicejuice, 
Quick question.  what is the update frequency? does the integration poll on a certain time frame?  i only ask as there don't seem to be many datapoints on the historical graph in HA?  thanks.  James

Lastaerie, 2023-05-16:
ah ok,.  Would it be possible to increase the timer on those two items?  not sure what a practical timer would be? 15 min? while an hour would seem to make sense for something that should be "static" when it develops a fault a shorter timing would give greater visibility.  many thanks  James

juicejuice, 2023-05-16:
Sure - the idea was that we don't need to hammer the API for data updates when the data changes infrequently. No one else has asked for these two data points so in this case it really depends on what you feel is appropriate for your situation. If it's 15 minutes then I can update that, no worries.

Lastaerie, 2023-05-16:
Its an interesting one right?  it's not something that normally would need updating as its supposed to be static but when errors/faults occur that's when you would like a more granular view.  pragmatically i think 15 would be fine as long as that doesn't cause any issues with API requests

juicejuice, 2023-05-16:
I think that 15 minutes works fine so I've made the change now, you should be able to update and check it out.

closed 2023-05-16 by juicejuice