juicejuice/homeassistant_redback #7: Current Load

Slade2000, 2023-05-05:
First, thank you very much, was waiting for this for a long time.

Is there a possibility for current load (Consumption) on the system. Currently only "Total Load" is available

thx

juicejuice, 2023-05-06:
Glad you found this valuable! I have wrestled with the same issue here, the Redback API doesn't provide the real-time power value for load and their support seem incapable of figuring out how to help. You can calculate this from available data though, something like:
`pv_output + battery_discharge - battery_charge + grid_export - grid_import`
Does that work as some sort of helper entity? Try it out and let me know? I could add a calculated entity into the integration if it works OK.

Slade2000, 2023-05-06:
I also went through the API's and thought it was just me missing the one
needed. Okay let me try out the calculations and I will revert back if I
get the right numbers.

juicejuice, 2023-05-07:
Thanks @Slade2000! I did a bit of messing around here and it looks like that calculation gets close to the figures from the Portal, but try as I might the values returned from the Redback API don't quite match those returned by the app or portal. I'm not sure if that is important to you or not. I would be interested to hear your own results?

Slade2000, 2023-05-11:
I have created a sensor with your calculations and it think it is pretty close. This definitely helps. What I am trying to achieve is, doing automation on pool pump etc. for when certain power usage vs generation, it then switch on or off.

juicejuice, 2023-05-11:
OK good feedback - I'll add this to the TODO list for next version. With any luck I will get some spare time this weekend.

juicejuice, 2023-05-12:
I've added a new "Site Load" sensor that is calculated from the other data points. Please test out my latest code commit @Slade2000 and let me know if that works for you now?

Slade2000, 2023-05-12:
Work perfect. Will keep an eye on how it goes. 

[image]

closed 2023-05-12 by juicejuice