juicejuice/homeassistant_redback #8: Phantom 8 minute AUTO mode Schedule constantly re appearing

LachyGoshi, 2023-06-28:
Hi Juice and Everyone,

Apologies for posting this here... I'm just at witts end with Installer and Redback support and thought that the community here might be of some assistance. Please let me know if there is a forum or an alternative place you recommend I post as this is NOT an issue with your integration.

My SH5000 system is stuck in Schedule Mode with an AUTO mode 8 minute long schedule phantomly applying itself. If i delete the schedule (either via the redback portal -> Settings page or via the DELETE schedule call on he swagger API page) another 8 minute long AUTO mode schedule re appears within a minute and re takes control of the sytem.
[image]
[image]
[image]
[image]

I have tried to quickly delete this schedule, then apply a set mode such as charge 600 watts, but by the time i navigate to the dashboard and back to settings sure enough, a new 8 minute long AUTO mode has appeared.

I have also tried adding other schedules over the top, but they just come and go and this active 8 minute long schedule that just seems to renew itself when its time is up, always seems to take priority.

Any ideas?

juicejuice, 2023-06-28:
Oh man that sounds brutal. I think the schedules are propagated from the Redback portal though, so I doubt it is something that can be fixed without the help of Redback engineering team. I don't know how you get access to them except via your installer and support.
I hope I'm wrong and someone else has a better idea!

Lastaerie, 2023-06-28:
Following the issues i had with my battery (Still awaiting the replacement..sigh) i used to send frequent emails to : customersupport@redbacktech.com / notifications@redbacktech.com ensuring my installer was also copied in tickets raised via the web portal just route directly to your installer which is about as much use a chocolate fireguard when the issue is with the portal and configuration.

I've also got the phone number : Redback Customer Support team : 1300 240 182 

Cheers 
James

GrantKeymer, 2023-06-29:
I also have an issue with one of my 4 batteries, going back a month now.  Redback have finally accepted that it is faulty, hence a warranty issue, but are taking their sweet time doing anything about it.  My installer is also about as much use as a chocolate fireguard 😠 

Because I'm in NZ, I don't think there would be any point in phoning their Customer Support Team.

LachyGoshi, 2023-07-15:
Ok so I did finally track down the cause of the phantom 8 minute auto schedule....

It was Evergen. A few weeks before the phantom schedules started appearing I had setup a free Evergen account to see if the app gave any easy control (either manually or if there were any user configurable smarts) but it couldn't - all it does is a basic read only overview of the system identical to the home dashboard page of the Redback portal.

Anyway, after a few weeks, Evergen decided to enable the "intelligent control" feature.... And even though I had no notification this feature was enabled, nor can you see it in the app that's it's activated, not can you disable it or in any way control your own system once it's activated. The idea is meant to be that you upload a copy of your bill (I didn't) so they know your tarifs and time of use hours etc, and then over time, let their AI smarts control your system for best savings/profit.

Evergen is all connected via their API link to redback. I did give Evergen my SH5000 serial number, but I honestly can't remember if I supplied them with my API credentials or not, but I don't think I did..... And the screenshots supplied by redback support had the username of the API calls as tech@evergen.com.au and I would have thought that if they were using my credentials then the redback logs would show my registered email address.

[image]

Anyway, now I new the API calls creating a continue 8 minute auto schedule were coming from Evergen, I was able to get onto their support and disable their "intelligent control".

I have not been able to confirm this, but if redback have authorised an open API to Evergen so that all Evergen need is a serial number and they can have total control of any Redback system then that is a huge security risk as in sure it wouldn't be difficult to just spoof serial numbers and get hits on valid/active units.

I have asked these questions but I do not expect an answer ... Perhaps one day I will go through the Evergen sign up process again to see if it does ask for any of my personal API credentials (and if they do.... Then joke is on me for being the fool).

Thanks all for the ideas on this anyway.

I'm still in pursuit of gaining easier full control of the Redback system and think somehow automating the API calls via home assistant in a way similar to Evergen with their non stop 8 minute long schedules will be the ultimate call, I'm just not smart enough to be able to figure it out and my head starts hurting when I think about how you would need to list all active schedules, delete them, and then load in a new schedule for whatever the command from home assistant is asking for!

closed 2023-07-15 by LachyGoshi