juicejuice/homeassistant_redback #11: Thanks & New Pull request

cabberley, 2023-07-06:
Hi JuiceJuice
Thanks for the great work in getting this integration up and running, something I have wanted to do for a long time.
I have made a pull request with some changes on your code adding a few more sensors I find useful.

I currently also have some HA functions that will set the Inverter/Battery to auto/charge/discharge & Rate. I will look at trying to add this into your repository, but the only way to set the active mode is through the "private API".

juicejuice, 2023-07-06:
Thanks @cabberley for the kind words. I'll checkout the PR this weekend!
If you make headway on the "private API" that might be interesting to add back in as it gives some additional data points?

LachyGoshi, 2023-10-25:
How's your merge of your charge/discharge control coming along @cabberley ? I would love this option but not cluey on how to code... Was thinking of trying to ask chat got but also not sure where to start! Let me know if you need a teatery!

cabberley, 2023-10-26:
> How's your merge of your charge/discharge control coming along @cabberley ? I would love this option but not cluey on how to code... Was thinking of trying to ask chat got but also not sure where to start! Let me know if you need a tester! I would also love to know how you got your current control of the battery running from HA and the API.

Have a look at this github repo to add what your looking for. With what Redback exposes, you can't change the inverter mode via the api unless you use schedules. This package uses the normal customer portal hooks I found to do it on demand via HA

https://github.com/cabberley/Home-Assistant_Redback_Invertor_Mode_Set

LachyGoshi, 2023-10-26:
thanks for the info - I'll take a look. I have been playing with the schedules in the API web page but it gets very messy very quickly!

LachyGoshi, 2023-11-28:
I got Caberlys pyscript based scripts working and I now have automation controls of the SH5000 battery system working well.jidt basic ones for now but can't wait to play more over summer with it.

As it was my first time with pyscript it wasnt the easiest install for a noob,  but i believe it would have been simpler if caberlys merge to Juices integration get approved and merged.

If juice merges caberlys changes, then caberlys scripts will automatically pick up the right serial numbers etc avoiding the issues and formats problems I experienced.

These are two great pieces of software that together are really powerful... Amazing that two guys have been able to come up with a better and more user friendly control interface than Redback themselves. Thanks so much.

juicejuice, 2024-06-27:
Closing this as @cabberley has now released a new integration with the control features.

closed 2024-06-27 by juicejuice