juicejuice/homeassistant_redback #12: Great work Chris Abberley and Juice 👏

cabberley, 2023-07-10:
Great work Chris Abberley and Juice 👏 

Is there any way we could access Battery SoH?  The website portal under 'System Details' provides a field for overall battery and individual battery SoH (4 fields in my case).  It looks like this...

[image]

For the past 6 weeks I've been waking up to this...

[image]

In early April, I noticed the first subtle changes to my charge and discharge curves which in hindsight, indicated something was not right with my batteries.  By the end of May, Battery 2 was discharging completely to about 2% whilst other batteries remained between 40 and 60%.  When Battery 2 is almost fully discharged, it cuts off the RS485 comms to all remaining batteries in the daisy chain, hence the message that my system has 'Missing Batteries 1/4'.

If I had been able to view the SoH for each battery, I suspect I would have noticed something amiss much earlier.  Battery 2 has now lost most of its capacity and I'm wondering if Battery 3 (the other 2.4kWh module) is headed down the same track.  Looking at the individual discharge curves, I can see that the 2.4kWh batteries are discharging more deeply each night than the other two 3.55kWh batteries.  This is probably why Battery 2 has given out first.

I stumbled across a Pylontech users page on Facebook and they were discussing access to the RS485 comms between batteries.  That is really inconvenient for me, but apparently it allows viewing all the individual battery statistics including SoH for each module.  Is there any way to access SoH via the Public or Private APIs?

_Originally posted by @GrantKeymer in https://github.com/juicejuice/homeassistant_redback/issues/10#issuecomment-1629754733_

cabberley, 2023-07-10:
@GrantKeymer 
https://github.com/GrantKeymer

Hi Grant, Thanks

The screen shots you have are not in the portal version that I have, and the data is not available from the APIs that RedBack provides us access to.

What is the website URI that you use to access the portal? there is an older v1 portal, but my account is only setup for v2.

Thanks

juicejuice, 2023-07-10:
@cabberley I can see the same screens as Grant, I remember not long after installation I asked my installer to unlock the "Analysis" page for my account. Once he did that then I could get access to all these extra details.

@GrantKeymer unfortunately the public API does not provide the data you are after. https://api.redbacktech.com/docs/index.html

I'm not going to spend any more time on the private API but if someone wants to have a go that would be interesting to see what extra datapoints can be retrieved!

GrantKeymer, 2023-07-11:
The screen with all the extra details has always been there in my system, and the URL is:

https://portal.redbacktech.com/systemdetails/SystemDetails?serialNumber=RBxxxxxxxxxxxxxx

It is under the 'Analysis' page as Juice mentioned above.  Some very valuable info in there and without it, I wouldn't have had a clue about what was happening with Battery 2.  I don't know whether my portal is v1 or v2 as it doesn't say that anywhere.

Sad to hear the Public API is so limited by comparison, but I understand your decision not to spend any more time on the Private API, when Redback could change it on a whim, at any time they like.

I'm just so grateful for all the work you've both done and looking forward to trying all these new sensors on my SH5000 single-phase system 😀

cabberley, 2023-07-11:
Thanks Grant
JuiceJuice did all the hard work, which is building the integration side out, I could never find any magic documentation to follow that made sense to me on HA doc sites.

There maybe an error when you install this with your single phase as it's not clear in the api doc what data gets returned if there is no phase 2 & 3 on your inverter, while the redbacklib correctly handles the return dataset the sensor.py assumes there will be 3 phases and will try and create entities for phase 2&3 and may error if redback api didn't return any data.

As for the "private"/portal APIs I agree with JuiceJuice, outside of the analysis page which I hadn't seen before the only thing it provides which is of value in my opinion is the dynamic changing of the inverter operation mode. Which I have built a seperate Pyscript repo to manage. 

Technically, the public API can set and update the schedule, but that's a whole new piece of work to have two way data flow between HA and Redback.

juicejuice, 2023-07-11:
@cabberley haha yes it took a bit of fiddling to understand how HA expects integrations to work. I'm still not 100% happy with the code but hey it does the job for now :) Good news though, others have tested and are using this integration with their single phases Redbacks and no problems.
BTW, can you access the system details directly via curl? curl -v --cookie ".AspNet.ApplicationCookie=COOKIE" https://portal.redbacktech.com/systemdetails/SystemDetails?serialNumber=SERIAL
@GrantKeymer it's a shame the public API is so limited but at least it has the basics. If someone in future decides to add private API support then we could perhaps access some of these extra details.

GrantKeymer, 2023-07-11:
@cabberley Thanks Chris.

My SH5000 has a 3-phase energy meter attached and I'm getting individual voltages for each phase, so hopefully all will be well with your new sensors.

I tried the API function which changes the inverter mode, but it said 'Access Denied' and I could never get any help from Redback to enable that function.  Their support really is abysmal, but with this integration, I'm getting so much more value from my system.  My EV charging algorithm is much smarter now, thanks to Juice 👏

cabberley, 2023-07-14:
> BTW, can you access the system details directly via curl? curl -v --cookie ".AspNet.ApplicationCookie=COOKIE" https://portal.redbacktech.com/systemdetails/SystemDetails?serialNumber=SERIAL

I haven't tried using curl, if you have a look at my code for setting the inverter you will find another header value you have to pass back

closed 2024-06-27 by juicejuice