juicejuice/homeassistant_redback #14: New Inverter Mode - Conserve

GrantKeymer, 2023-08-03:
Good morning guys,

A couple of weeks ago, my faulty Pylontech battery was replaced and I now have enough capacity to get through the night again - wonderful 👍 

I've just noticed there is a new Inverter Mode available in Settings.  It's called Conserve.

Have either of you tried it, and if so, what does it do?

cabberley, 2023-08-03:
It's not new, not sure why you haven't seen it before. Conserve mode is supposed to conserve your battery, so if Pv is greater than your household consumption, then it will charge battery or fit if battery is 100%.
If pv is not enough for household load it will import from grid and not use battery.

I have a list somewhere that explains all the modes. There are several which aren't exposed as well, but they aren't really any use.
If I find it I will pose it here later.

GrantKeymer, 2023-08-03:
Great - thanks Chris.  That sounds exactly how I'd prefer my inverter to be working all the time, so I'll give it a try.

Recently I made use of the Batt Discharge sensor to ensure that my EV charger wasn't discharging the house battery in the late afternoon, so this will hopefully be another step along the way to getting my system working just the way I want it.

cabberley, 2023-08-03:
[image]

GrantKeymer, 2023-08-04:
Awesome, many thanks Chris, that is very helpful.

I've since tried 'Conserve' mode and realised it will import grid power at any time solar power is available, if it finds Site Load > Solar Generation.  I would rather it waited until the cloud passed and replenished the battery, rather than importing from the grid to resolve a temporary situation.

So I'll probably stick with 'Auto' mode but will study the table you've posted, in case there's a better option.

cabberley, 2023-08-04:
I haven't tested it too much but using the import power setting would probably enable pv curtailment control during negative fit of battery is full

LachyGoshi, 2023-08-13:
Disclosure: this is based on my testing of my sh5000: Just be cautions, as my understanding is that the "Import Limit" sets the overall power limit for the overall system, not a limit or target specifically of GRID import. 

For example, on my sh5000, if I enable Import Limit and set it to 2000w,  assuming my battery is flat, at night this would typically be importing 2000w from the grid and charge the battery at about 1500w and also cover the house load of about 500w. If house load exceeds 2000w then it just all imports from the grip for consumption and not charge the battery. During the day, when the sun is out, this still only charges the battery at about 1500w and covers my house load of 500w, difference being this is coming from the PV panels (not the grid). So that 2000w limit is like a system wide limit, and it's the same on "Export Limit" mode also. So there's a huge wast of PV energy that isn't being utilised (eg my panels could be generating 5000w but due to the Import/Export Limit it's only utilising 2000w.)

I think the what redback call import and export is very misleading as it should relate to the GRID and not their own interlan system.

GrantKeymer, 2023-08-14:
I have also been experimenting with the various inverter modes and found some of the mode descriptions at best misleading and at times counterintuitive regarding the way they operate.  About 2 years ago, I was attempting to prevent wastage of solar power whilst our underfloor heating was running for part of the morning.  Here are some notes I made back then...

It's a bit more complicated in the morning, when there is some solar power available for at least part of the period between 7:30am and 10am, while the underfloor heating is running.

After some experimenting, it seems as though the inverter can be programmed to avoid exporting solar power to the grid to offset power being consumed by the underfloor heating on a different phase.  Using the mode 'Import Power' with a rate of 0 Watts effectively diverts any solar power directly to the battery, unless the battery is unable to cope with all of it.  In that case, the excess solar power is used to offset power being imported from the grid for household loads.  Otherwise, all power consumed comes from the grid, with all solar power going into the battery.  This arrangement continues until 10am when the underfloor heating switches off and 'Auto' mode resumes.

A mode of 'Export Power' with a rate of 0 Watts was tried, but it effectively shut down the solar arrays with generation being limited to about 100 Watts.  A mode of 'Import Power' with a rate of 500 Watts seemed the logical solution for the morning underfloor heating period, but it caused unexpected behaviour with around 1000 Watts of additional power being imported to charge the battery.  Therefore, using 'Import Power' together with a rate of 0 Watts has proven to be the most useful, when intuition would suggest otherwise.

Back then, 'Conserve' mode wasn't available, so to prepare for a severe storm, together with the possibility of Grid Outages, I used to select 'Charge Battery' mode with a rate of 1000 Watts.  This was done as soon as possible after receiving a severe weather warning, so our battery would be kept fully charged at all times, with all necessary power being imported from the grid to keep it that way.  After the storm had passed, I returned to 'Auto' mode.

Now it seems, the better option would be to use 'Conserve' mode, so only PV power is used to charge the battery, rather than importing as I've done in the past.  Of course, this assumes there is sufficient time for that to happen, with a decent amount of sunshine being available.  Otherwise, 'Charge Battery' mode will still be the best way to prepare for an imminent storm.

closed 2023-08-31 by GrantKeymer