juicejuice/homeassistant_redback #15: Energy Tab only allowing "total" entities to be chosen.

Vipesss, 2023-08-30:
Hello! Recently installed the integration and am noticing some odd things. 

I have the SH5000 inverter, everything looks to be coming through as expected how when I go to the energy tab and begin inputting those details the entity field only lets me choose entities that end in "total".

For example, Grid Consumption allows me to choose "Grid Import Total". The same with all other fields. I can find the relevant entity but only its total value which makes the energy tab display incorrect information. 

Am I doing something wrong? 

Cheers,
Denis
[image]

juicejuice, 2023-08-30:
That is as expected. The Redback integration provides two types of sensors: power and energy. Power is an instantaneous reading, whereas energy is a volume that is a measurement of power over a period of time. Power is commonly expressed in kilowatts (kW), whereas energy is commonly measured in kilowatt-hours (kWh).
By way of example, if my house used 3kW of power over one hour then that is 3kWh of energy. If my house only used 1kW of power but over 3 hours, that is also 3kWh.
The Home Assistant dashboard does not support power readings, only energy measurements. That is the main reason I added the total sensors, so they could provide energy measurements for the HA dashboard to use.
I'm curious what makes you suspect the dashboard is showing incorrect information? I've used this integration for a long time now and it is fairly accurate as compared to Redback's app and web portal.

Vipesss, 2023-08-31:
Hey Juice, 

Thanks very much for the quick response. Absolutely understand where you're coming from. This is likely where I am misunderstanding the concept of the energy tab so ill try to break down further. 

By using the total entities the information returned is telling me the total amount for that day (i presume). Where it's mainly bothering me is on the Energy Distribution graph. For example the below is a snapshot of both the Energy Distribution graphic in HA and the RedBack portal. I have not found a way to set these to anything other that the "total"entity. Meaning its showing my total generated solar for that day rather than the amount generated as can be found in the RedBack Solar Generation entity.
[image]

[image]

I have set up an energy dashboard separately to the built in one where I can reference the individual entities and it works a treat. Really just wanted to understand if I was doing something wrong here.

Cheers,
Denis

juicejuice, 2023-08-31:
Gotcha! The HA energy dashboard does not support instantaneous power readings, as you say it will show totals according to your selection at top of screen (day, week, month, year). You would have to build your own custom dashboard to replicate what Redback's portal shows. Actually, that could be a fun project if someone hasn't done it already?

Vipesss, 2023-08-31:
Hey Juice, 

Thanks for confirming mate. I just wanted to make sure I didn't have something backwards. Good to know its working as expected. I shall have a bit of a play around to see how i can configure it for my needs. 

Thanks again and appreciate the effort to put this integration together. 

Cheers,
Denis

closed 2023-08-31 by Vipesss