juicejuice/homeassistant_redback #16: No Configuration after installation

Ross-FS, 2023-09-15:
I have installed the Redback Integration into my HomeAssistant system. It works fine. However, I am not able to view or change the Client ID and Credentials now that it is running.
There is no option to change the configuration from within the settings after it has been installed. This is a bit of a short-coming, as I am now unable to review these details for troubleshooting, or migration.
It appears, that if the Redback company changes my security details, then I would have to delete and recreate this integration.
Would you be able to add a Configuration command to the integration?

juicejuice, 2023-09-17:
Thanks, great idea! I will look into it when I get chance. Unfortunately, I'm away from home until October.

GrantKeymer, 2023-11-22:
Hi @juicejuice,

Today I received a notification email from Redback that my API access will expire in 2 months time.  Apparently this now happens every 2 years.  Is there any chance of having a Configuration Flow available and working within the next 2 months?  I really don't want to have to reinstall the integration, when it's been working so well for so long.

Secondly, Redback have provided a Web Link to my new credentials, which will expire in 30 days.

If I access the link to my new credentials, is that likely to disable my existing credentials?

juicejuice, 2023-11-23:
Interesting! I haven't had a similar email yet, so I don't know if it replaces old with new credentials.
I'll look into what's involved to change credentials, I've been flat out with work but hopefully get some spare time soon. In the meantime, you can manually update the details if required:
1. Login to HA and launch File editor add-on
2. Click folder button in top-left to open the /config/.storage/core.config_entries file
3. Search in that config file for "redback"
4. Update the following "client_id" and "auth" fields to suit new credentials
5. Click red save button at top of File editor
6. Reload the Redback integration (or restart HA)

Here is what the config entry code should look like:
`"domain": "redback",`
`"title": "Redback",`
`"data": {`
`    "displayname": "Redback",`
`    "apimethod": "public",`
**`    "client_id": "0123456a-1111-1111-1111-1a2b3c4d5e6f",`**
**`    "auth": "ABCDE~XCC4qulr-JzXBVVWQRVcnXrBO8N_EcZbrv",`**
`    "site_index": "First"`
`},`

Note: if you have multiple inverters registered you will need to update every instance of "redback" in that config file.

GrantKeymer, 2023-11-23:
Brilliant, thanks very much Juice.  The manual method sounds very simple, so I'll give it a try.

GrantKeymer, 2023-11-23:
First thing to note is that issuance of new credentials doesn't affect the validity of the old ones.  In fact, the clock was already ticking on my new credentials, from the date they were made available.

juicejuice, 2023-11-26:
OK so I've just pushed an update to the integration. It turns out Home Assistant has a standard "re-authentication" mechanism that other integrations also use. I've added this re-authentication support for the Redback integration now. The way it works is to detect when credentials are expired and notify you of the issue. You then have the option to supply new credentials and it will test that those are working before updating your configuration. I've tested out here and it is working fine, please check it out and let me know? (BTW, I haven't found a way to manually trigger re-authentication except for editing the core.config_entries file by hand with incorrect credentials)

GrantKeymer, 2023-11-26:
I just had a go at editing the core.config_entries file and didn't get very far.  Following your instructions...

`Login to HA and launch File editor add-on
Click folder button in top-left to open the /config/.storage/core.config_entries file
`

For some reason, the File Editor cannot see the **.storage** directory in the config directory.  I'm guessing it's because there is a '.' in front of it.  However, I can't see any way to change that behaviour.

Using Terminal, I can successfully navigate to the .storage directory and using the Cat command, I can view the contents of the file and see the **redback** section.

I guess I could use Vim or Nano to edit the file, but the File Editor is so much easier.  Any idea on how I could make it work?

I am running Home Assistant OS, rather than Core, so maybe that has something to do with it.

juicejuice, 2023-11-26:
Check the configuration tab of the File Editor add-on. It has an "ignore" setting which might be what is hiding your storage folder. Looks like there is a thread here on the subject too: https://community.home-assistant.io/t/access-storage-folder/190186/18

GrantKeymer, 2023-11-26:
Many thanks once again Juice, for your excellent support and clear instructions.

As you suggested, the .storage folder was hidden because of an entry in the ignore setting.  I deleted that entry, along with 3 others, turned off another switch which was restricting me to the config folder, and voila!, there was .storage.  Now I can navigate anywhere in the file system.  No doubt, the authors of the File Editor add-on did this to protect newbies from tinkering with stuff they shouldn't be tinkering with.

Upon restarting HA, I was prompted to download an updated version of your Redback integration, then prompted for the new Configuration (credentials), and finally needed to restart HA again to allow the new credentials to take effect.

After a short delay, all the Redback entities reappeared as they were before.

Great work Juice!  I will save these instructions for use when my new credentials expire in 2 years time 👍 

Meanwhile, I hope this thread will be helpful to others when their credentials expire.

closed 2023-11-26 by juicejuice