juicejuice/homeassistant_redback #17: Float type error in sensor.py

LachyGoshi, 2023-10-26:
I get this error in the home assistant logs:

Log details (ERROR)
This error originated from a custom integration.
Logger: homeassistant
Source: custom_components/redback/sensor.py:467
Integration: Redback Technologies Portal ([documentation](https://github.com/juicejuice/homeassistant_redback), [issues](https://github.com/juicejuice/homeassistant_redback/issues))
First occurred: 9:29:26 PM (2 occurrences)
Last logged: 9:30:25 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 233, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 389, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 172, in async_update_listeners
    update_callback()
  File "/config/custom_components/redback/sensor.py", line 467, in _handle_coordinator_update
    measurement = float(eval(measurement, {'ed':ed}))
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 1, in <module>
TypeError: float() argument must be a string or a real number, not 'NoneType'

I'm sure this is a simple correction if someone can assist?

juicejuice, 2023-10-29:
I don't see that error in my logs here, so there must be something interesting going on. That bit of code only runs for the "Site Load" sensor, so I'm assuming your Site Load will be a missing value?
Here is what it is trying to do:
`"data_source": "$calc$ float(ed['PvPowerInstantaneouskW']) + float(ed['BatteryPowerNegativeIsChargingkW']) - float(ed['ActiveExportedPowerInstantaneouskW']) + float(ed['ActiveImportedPowerInstantaneouskW'])",`
So, basically, add the PV power to Battery power, subtract exported power, add back in the imported power.
Quick first question, do you have a battery in your system or not?

juicejuice, 2023-11-26:
@LachyGoshi I just realised I hadn't heard back from you on this one. Did you get chance to look into this any further?

LachyGoshi, 2023-11-27:
Hey mate, sorry I missed your last one and been flat out travelling for work allot. Back home this weekend so will have a proper look. Sorry for the delay and thanks for the follow up.

LachyGoshi, 2023-11-28:
Hi Juice,

The error does remain, exactly the same entry in the logs after start up.

As I run two inverters, the SH5000 (with a battery) is fine. This error is coming from the SI5000.

As I get the site load ok from the SH5000 for now I have just disabled the site load entity for the SI5000. This hasn't fixed the issue, but I'm just ignoring it and I no longer get the log entry because the entity isn't being loaded at start up any more.

I'm happy to help test any fix if anyone else out there has same issue and fix is released, just reach out and let me k ow, but for now, I'm happy with is disabled as I have the other one working on.

juicejuice, 2023-11-28:
Thanks for the confirmation! It is most likely what I thought then, that it is trying to use a battery value that doesn't exist. I'll have a look into fixing that on the weekend.

juicejuice, 2023-12-14:
@LachyGoshi my latest commit should fix this error message now and restore the Site Load sensor.

closed 2023-12-14 by juicejuice

LachyGoshi, 2023-12-14:
Yep. looks good thanks as always @juicejuice ! Clean logs on start up now!