juicejuice/homeassistant_redback #18: Battery count in API is misleading

Mark-Hetherington, 2023-12-13:
After getting API credentials, setting up the Redback integration was straightforward, and I started getting meaningful data straight away. However I'm not seeing anything about my battery. Looking at the code I can see that it quite reasonably looks at the `/Api/v2/EnergyData/{siteId}/Static`, and then looks at the `Data.StaticData.Nodes[0].StaticData.BatteryCount` to see if the system has a battery. In my case this is returning "0":
```
"Nodes": [
      {
        "StaticData": {
          "ModelName": "SH5000",
          "BatteryCount": 0,
          "SoftwareVersion": "1.5.17.37",
          "FirmwareVersion": "15157",
          "BatteryModels": [],
          "Id": "RB17121802200037",
          "Type": "Inverter",
          "DynamicDataMetadata": null
        },
        "Nodes": null
      },
...
]
```
However we definitely have a battery, listed in the web portal. 
[image]

Also the `/Api/v2/EnergyData/{siteId}/Dynamic` endpoint returns data including:
```
"BatterySoCInstantaneous0to1": 0.98,
    "PvPowerInstantaneouskW": 1.841,
    "InverterTemperatureC": 48.6,
    "BatteryPowerNegativeIsChargingkW": 0.017,
```
I think I can just rework this to look at the dynamic information to confirm if there is a battery, but I think this is really a bug on Redbacks side. I'll reach out to their customer support while I'm looking at any code changes.

juicejuice, 2023-12-14:
Yes we can only work with the data that is available. I think the BatteryCount and BatteryModels are details that the installer is meant to supply when they complete system installation. They are meant to take photos of everything and document the as-installed configuration of equipment. This static data is all registered with the Redback central database.
The dynamic data, on the other hand, appears to be the live production data captured by the inverter. I guess this could be used but it doesn't provide all details - for example, the 9.1kWh battery capacity field shown in your screenshot.
I'll have a look at your PR.

closed 2023-12-14 by juicejuice

Mark-Hetherington, 2023-12-14:
Ok, so perhaps this isn't so much a bug, as a missed step in the installation. I wonder if I can get Redback to fix this data. I guess I'll see what their response is.

Our inverter was replaced after lightning damage, so perhaps it hasn't all come across correctly. We have a few other odd symptoms, like our export limit being lost if the inverter control board powers off.

Mark-Hetherington, 2023-12-14:
Also, I should apologise. I neglected to take the time to thank you for the effort that has gone into crafting this integration. As I get more familiar with the code I'm liking the way it's all put together.

juicejuice, 2023-12-14:
> Ok, so perhaps this isn't so much a bug, as a missed step in the installation. I wonder if I can get Redback to fix this data. I guess I'll see what their response is.

Good idea!

> Also, I should apologise. I neglected to take the time to thank you for the effort that has gone into crafting this integration. As I get more familiar with the code I'm liking the way it's all put together.

Thanks for the kind words. It was a fun project to put together and has been very helpful for my house. I learned a lot about HA and Python during the work. I'm not super proud of the code itself, it's not so elegant, but it does the job and seems reliable and maintainable.

Mark-Hetherington, 2024-07-01:
I have not had a meaningful response from Redback customer support. So this is mainly an update to say there isn't likely to be further developments on this.

juicejuice, 2024-07-01:
Sorry to hear that @Mark-Hetherington. If Redback can't help then I think the only other option would be for your installer to update the battery configuration. They get special access to update such things which should refresh the database records at Redback.