juicejuice/homeassistant_redback #23: Invalid authentication - but credentials work at https://api.redbacktech.com

fhopley, 2024-02-21:
Hi,

I've recently got API credentials (Client ID & credential) from Redback and I'm attempting to use the homeassistant_redback custom component.  I've installed it via HACS, restarted, and tried to add the integration.  When I put in the supplied Client ID as the Redback ID, and the credentials as the Redback Authentication - I get an 'Invalid authentication' error.  However, when I use the same Client ID and credentials at the API test site (https://api.redbacktech.com) I get a successful authentication / login (see attached image of login result).  I can't see anything in the logs associated with redback after the attempt to add / configure the integration - apart from the warning at boot that 'redback' it is a custom integration.

Please let me know if you can think of anything for me to try get the integration to work.

[image]

juicejuice, 2024-02-23:
This sounds like the same issue someone else had last year: https://github.com/juicejuice/homeassistant_redback/issues/4
Or is this different?

fhopley, 2024-02-23:
Hi,Yes - this is _exactly_ the same issue. I will continue trying to get Redback support to give me credentials that work properly on the api test site. My guess is that they have created the credentials but not associated them with any site / nmi / inverter. Thanks,Fred Hopley0423 029 145

fhopley, 2024-02-25:
I'm still trying to get information out of Redback support, but one thing that came to mind is that my 'user access' account is set up as 'Home User Monitor / Inspector' rather than 'Home User All Access'.  Let me know if you think this might be a potential cause.

cabberley, 2024-02-25:
On the API site can you retrieve your site ID and then use that for subsequent APi calls and see your data?

fhopley, 2024-02-25:
No - the authentication works, but all API calls except for Alive fail.  For example see /Api/v2/EnergyData for getting available sites example attached.  I've tried all the API calls that I can including those that require either the serial number or the NMI, but none work. 

[image]

juicejuice, 2024-02-25:
Yes it does sound like Redback has not configured your account correctly. I don't think you can do anything except work with their support team. Once they get your API access working correctly then you will be able to use this integration.

closed 2024-06-27 by juicejuice