juicejuice/homeassistant_redback #25: Multiple inverters

homeskool2, 2024-05-20:
hey 

love the app but i can only seem to get one of my inverters to show up. 

I have 2 at home working in what im told is a master and worker configuration, i get an unexpected error when trying to add a second device.

I have tested from the website and both serials number return a response

any ideas on what might be causing the issue?

juicejuice, 2024-05-20:
We definitely have had people with two inverters get this working. Did you set the "Redback Site" field properly when adding the inverters? You're meant to select "First" for first inverter and then add Redback again and select "Second" for the second inverter. If that is producing an error then I guess we'll need to see the error messages from the HA log.

homeskool2, 2024-05-21:
no matter what option i pick for the inverter i get the following in the logs

Logger: custom_components.redback
Source: custom_components/redback/config_flow.py:93
integration: Redback Technologies Portal ([documentation](https://github.com/juicejuice/homeassistant_redback), [issues](https://github.com/juicejuice/homeassistant_redback/issues))
First occurred: 9:06:20 PM (2 occurrences)
Last logged: 9:07:57 PM

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/redback/config_flow.py", line 93, in async_step_user
    self._abort_if_unique_id_configured()
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2165, in _abort_if_unique_id_configured
    raise data_entry_flow.AbortFlow(error)
homeassistant.data_entry_flow.AbortFlow: Flow aborted: already_configured

juicejuice, 2024-05-21:
OK, wonderful, thank you. I don't have multiple inverters so it is hard for me to test. It is possible that when I added the auto-reconfigure support it has messed up the process for adding multiple inverters. I'll look into it.

homeskool2, 2024-05-21:
would it be easy to add a serial number checker or something to tell the difference between the two systems?

juicejuice, 2024-05-21:
OK what it is using in the code right now is the "site ID" as the unique ID for each inverter. You mentioned a master/worker system, I'm not sure we've seen that before. If each inverter doesn't have a unique Site ID then that is the problem. For example, they might share a Site ID but have unique serial numbers.
To help with diagnosis, can you please send some data samples? I will need to review the JSON data files. From Redback's Swagger API site (https://api.redbacktech.com/docs/index.html) you need to retrieve two EnergyData samples, one for Static and one for Dynamic data. Please sanitise those samples (they may contain some personal info) then send them over. That should be enough for me to check against.

homeskool2, 2024-05-22:
i can confirm that they both share the same site id

maybe this api would work better? 
https://api.redbacktech.com/Api/v2/EnergyData/Static/BySerialNumber/

STATIC
{
  "Data": {
    "StaticData": {
      "TimestampUtc":       },
      "SiteDetails": {
        "GenerationHardLimitVA": null,
        "GenerationSoftLimitVA": null,
        "ExportHardLimitkW": null,
        "ExportHardLimitW": null,
        "ExportSoftLimitkW": null,
        "ExportSoftLimitW": null,
        "SiteExportLimitkW": null,
        "BatteryMaxChargePowerkW": 4.95,
        "BatteryMaxDischargePowerkW": 4.95,
        "BatteryCapacitykWh": 11.904,
        "UsableBatteryCapacitykWh": 10.714,
        "PanelModel": " ",
        "PanelSizekW": 9.62,
        "SystemType": "Hybrid",
        "InverterMaxExportPowerkW": 5,
        "InverterMaxImportPowerkW": 5
      },
      "CommissioningDate": "2021-01-08",
      "NMI": "QB016680721",
      "LatestDynamicDataUtc": "2024-05-22T07:52:21Z",
      "Status": "OK",
      "Id": "S4d2f2a******",
      "Type": "Site",
      "DynamicDataMetadata": {
        "ActiveExportedPowerInstantaneouskWMetadata": {
          "Measured": true
        },
        "ActiveImportedPowerInstantaneouskWMetadata": {
          "Measured": true
        },
        "VoltageInstantaneousVMetadata": {
          "Measured": true
        },
        "CurrentInstantaneousAMetadata": {
          "Measured": true
        },
        "PowerFactorInstantaneousMinus1to1Metadata": {
          "Measured": true
        },
        "FrequencyInstantaneousHzMetadata": {
          "Measured": true
        },
        "BatterySoCInstantaneous0to1Metadata": {
          "Measured": false
        },
        "PvPowerInstantaneouskWMetadata": {
          "Measured": false
        },
        "InverterTemperatureCMetadata": {
          "Measured": true
        },
        "BatteryPowerNegativeIsChargingkWMetadata": {
          "Measured": false
        },
        "PvAllTimeEnergykWhMetadata": {
          "Measured": false
        },
        "ExportAllTimeEnergykWhMetadata": {
          "Measured": true
        },
        "ImportAllTimeEnergykWhMetadata": {
          "Measured": true
        },
        "LoadAllTimeEnergykWhMetadata": {
          "Measured": false
        }
      }
    },
    "Nodes": [
      {
        "StaticData": {
          "ModelName": "SH5000",
          "BatteryCount": 4,
          "SoftwareVersion": "2.19.32310.1",
          "FirmwareVersion": "252519",
          "BatteryModels": [
            "US3000A",
            "Unknown",
            "Unknown",
            "Unknown"
          ],
          "Id": "RB20061402210***",
          "Type": "Inverter",
          "DynamicDataMetadata": null
        },
        "Nodes": null
      },
      {
        "StaticData": {
          "Id": "HouseLoad",
          "Type": "Houseload",
          "DynamicDataMetadata": null
        },
        "Nodes": null
      }
    ]
  }
}

DYNAMIC

{
  "Data": {
    "TimestampUtc": "2024-05-22T07:54:21Z",
    "SiteId": "S4d2f2a2****",
    "Phases": [
      {
        "Id": "A",
        "ActiveExportedPowerInstantaneouskW": 0.404,
        "ActiveImportedPowerInstantaneouskW": 0,
        "VoltageInstantaneousV": 240.8,
        "CurrentInstantaneousA": 5.81,
        "PowerFactorInstantaneousMinus1to1": 0.028
      },
      {
        "Id": "B",
        "ActiveExportedPowerInstantaneouskW": 0.03,
        "ActiveImportedPowerInstantaneouskW": 0,
        "VoltageInstantaneousV": 240.6,
        "CurrentInstantaneousA": 1.08,
        "PowerFactorInstantaneousMinus1to1": 0.127
      },
      {
        "Id": "C",
        "ActiveExportedPowerInstantaneouskW": 0.958,
        "ActiveImportedPowerInstantaneouskW": 0,
        "VoltageInstantaneousV": 241.1,
        "CurrentInstantaneousA": 4.21,
        "PowerFactorInstantaneousMinus1to1": 0.818
      }
    ],
    "FrequencyInstantaneousHz": 50.01,
    "BatterySoCInstantaneous0to1": 0.84,
    "PvPowerInstantaneouskW": 0,
    "InverterTemperatureC": 26.6,
    "BatteryPowerNegativeIsChargingkW": 0.314,
    "PvAllTimeEnergykWh": 38911.3,
    "ExportAllTimeEnergykWh": 29565.53,
    "ImportAllTimeEnergykWh": 15307.61,
    "LoadAllTimeEnergykWh": 45087.2,
    "Status": "OK",
    "Inverters": [
      {
        "SerialNumber": "RB200614022****",
        "PowerMode": {
          "InverterMode": "Auto",
          "PowerW": 0
        }
      }
    ]
  },
  "Metadata": {
    "Latest": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic?metadata=True",
    "Permalink": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/At/20240522T075421Z?metadata=True",
    "Back": {
      "1m": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240522T075322Z/1?metadata=true",
      "10m": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240522T074422Z/2?metadata=true",
      "1h": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240522T065422Z/5?metadata=true",
      "1d": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240521T075422Z/60?metadata=true",
      "1w": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240515T075422Z/60?metadata=true"
    },
    "Forward": {
      "1m": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240522T075522Z/1?metadata=true",
      "10m": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240522T080422Z/2?metadata=true",
      "1h": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240522T085422Z/5?metadata=true",
      "1d": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240523T075422Z/60?metadata=true",
      "1w": "/Api/v2/EnergyData/S4d2f2a25b5e3/Dynamic/LatestBeforeUtc/20240529T075422Z/60?metadata=true"
    }
  }
}

juicejuice, 2024-05-22:
Sorry mate I need one other data point from the API, if you would be so kind? I need the output from **/Api/v2/EnergyData/With/Nodes** , that is what is used to enumerate the available inverters. Please don't forget to sanitise the output so you are not sharing any sensitive details.

homeskool2, 2024-05-22:
no problem

{
  "Page": 0,
  "PageSize": 100,
  "PageCount": 2,
  "TotalCount": 2,
  "Data": [
    {
      "Id": "S4d2f2a******",
      "Nmi": "QB0166******1",
      "Type": "Site",
      "Nodes": [
        {
          "SerialNumber": "RB20061402******3",
          "Id": "RB2006140******3",
          "Nmi": "QB0166******1",
          "Type": "Inverter",
          "Nodes": null
        },
        {
          "Id": "Houseload",
          "Nmi": "QB0166******1",
          "Type": "Houseload",
          "Nodes": null
        }
      ]
    },
    {
      "Id": "S4d2f2a******",
      "Nmi": "QB0166******1",
      "Type": "Site",
      "Nodes": [
        {
          "SerialNumber": "RB2006140******7",
          "Id": "RB2006140******7",
          "Nmi": "QB0166******1",
          "Type": "Inverter",
          "Nodes": null
        },
        {
          "Id": "Houseload",
          "Nmi": "QB0166******1",
          "Type": "Houseload",
          "Nodes": null
        }
      ]
    }
  ]
}

juicejuice, 2024-05-22:
In your original JSON data, both of the "Id" fields are identical values, right? If so, that is the issue. All other users to date have had unique Site IDs for each inverter. That is why it won't let you configure a second inverter, as it doesn't think it is a separate device. I can't remember if there was a reason we didn't use serial numbers or not. It may be possible to switch, I'll have to check into it.

homeskool2, 2024-05-22:
correct the it fields are the same for both

juicejuice, 2024-05-24:
I've had a look and I think I can update this to use serial number as the unique ID for Redback devices in HA. I'll reach out after I've had chance to work on it so you can test out.

homeskool2, 2024-05-24:
awesome

cabberley, 2024-05-26:
Hi @homeskool2 could you provide a couple more samples please? 
Curious to see if both inverters show up in this response.
**https://api.redbacktech.com/Api/v2/EnergyData/{siteid}/Static**

from this API when you put both Serial numbers in the request please?
**https://api.redbacktech.com/Api/v2/EnergyData/Multiple/BySerialNumber/Dynamic**

Thanks

homeskool2, 2024-05-26:
hey on the static one only one inverter shows up

{
  "Data": {
    "StaticData": {
      "TimestampUtc": "2024-05-26T21:45:49.4391983Z",
      "Location": {
        "Latitude": -xxxxx,
        "Longitude": xxxxx,
        "AddressLineOne": "xxxxx",
        "AddressLineTwo": null,
        "Suburb": "xxxxx",
        "State": "QLD",
        "Country": "Australia",
        "PostCode": "xxxxx"
      },
      "TechnologyProvider": "RBT",
      "RemoteAccessConnection": {
        "Type": "ETHERNET",
        "CustomerChoice": true
      },
      "ApprovedCapacityW": null,
      "SolarRetailer": {
        "Name": "Energy Solution Centre",
        "ABN": "84163501488"
      },
      "SiteDetails": {
        "GenerationHardLimitVA": null,
        "GenerationSoftLimitVA": null,
        "ExportHardLimitkW": null,
        "ExportHardLimitW": null,
        "ExportSoftLimitkW": null,
        "ExportSoftLimitW": null,
        "SiteExportLimitkW": null,
        "BatteryMaxChargePowerkW": 4.95,
        "BatteryMaxDischargePowerkW": 4.95,
        "BatteryCapacitykWh": 11.904,
        "UsableBatteryCapacitykWh": 10.714,
        "PanelModel": " ",
        "PanelSizekW": 9.62,
        "SystemType": "Hybrid",
        "InverterMaxExportPowerkW": 5,
        "InverterMaxImportPowerkW": 5
      },
      "CommissioningDate": "2021-01-08",
      "NMI": "QB0166xxxxx",
      "LatestDynamicDataUtc": "2024-05-26T21:45:21Z",
      "Status": "OK",
      "Id": "S4d2f2axxxxx",
      "Type": "Site",
      "DynamicDataMetadata": {
        "ActiveExportedPowerInstantaneouskWMetadata": {
          "Measured": true
        },
        "ActiveImportedPowerInstantaneouskWMetadata": {
          "Measured": true
        },
        "VoltageInstantaneousVMetadata": {
          "Measured": true
        },
        "CurrentInstantaneousAMetadata": {
          "Measured": true
        },
        "PowerFactorInstantaneousMinus1to1Metadata": {
          "Measured": true
        },
        "FrequencyInstantaneousHzMetadata": {
          "Measured": true
        },
        "BatterySoCInstantaneous0to1Metadata": {
[... 282 lines cut ...]
          "SoftwareVersion": "2.19.32310.1",
          "FirmwareVersion": "252519",
          "BatteryModels": [
            "US3000A",
            "Unknown",
            "Unknown",
            "Unknown"
          ],
          "Id": "RB20061402xxxxxx",
          "Type": "Inverter",
          "DynamicDataMetadata": null
        },
        "Nodes": null
      },
      {
        "StaticData": {
          "Id": "HouseLoad",
          "Type": "Houseload",
          "DynamicDataMetadata": null
        },
        "Nodes": null
      }
    ]
  }
}

cabberley, 2024-05-27:
Thanks for that. Also did you manage to get the results from this?

from this API when you put both Serial numbers in the request please?

https://api.redbacktech.com/Api/v2/EnergyData/Multiple/BySerialNumber/Dynamic

Thanks

homeskool2, 2024-05-27:
That one is asking for a string to do the request not the serial number. Do you have an idea what the string would be?

juicejuice, 2024-05-27:
> That one is asking for a string to do the request not the serial number. Do you have an idea what the string would be?

Put in both serial numbers, something like ["RB11111111", "RB22222222"]

juicejuice, 2024-05-27:
> hey on the static one only one inverter shows up

Oh dear, that's going to be a bigger change then. It means we would have to rewrite the data gathering functions to use serial numbers for everything as Site ID is no longer relevant.

homeskool2, 2024-05-27:
Oh that simple lol

{
  "TotalCount": 2,
  "Data": [
    {
      "TimestampUtc": "2024-05-27T04:50:21Z",
      "SiteId": "S4d2f2axxxxx",
      "Phases": [
        {
          "Id": "A",
          "ActiveExportedPowerInstantaneouskW": 2.432,
          "ActiveImportedPowerInstantaneouskW": 0,
          "VoltageInstantaneousV": 241.1,
          "CurrentInstantaneousA": 10.63,
          "PowerFactorInstantaneousMinus1to1": 0.968
        },
        {
          "Id": "B",
          "ActiveExportedPowerInstantaneouskW": 3.051,
          "ActiveImportedPowerInstantaneouskW": 0,
          "VoltageInstantaneousV": 240.8,
          "CurrentInstantaneousA": 12.7,
          "PowerFactorInstantaneousMinus1to1": 0.995
        },
        {
          "Id": "C",
          "ActiveExportedPowerInstantaneouskW": 3.564,
          "ActiveImportedPowerInstantaneouskW": 0,
          "VoltageInstantaneousV": 241.3,
          "CurrentInstantaneousA": 14.79,
          "PowerFactorInstantaneousMinus1to1": 0.998
        }
      ],
      "FrequencyInstantaneousHz": 49.97,
      "BatterySoCInstantaneous0to1": 0.95,
      "PvPowerInstantaneouskW": 0.543,
      "InverterTemperatureC": 34.7,
      "BatteryPowerNegativeIsChargingkW": 1.323,
      "PvAllTimeEnergykWh": 39034.9,
      "ExportAllTimeEnergykWh": 29770.54,
      "ImportAllTimeEnergykWh": 15477.26,
      "LoadAllTimeEnergykWh": 45434,
      "Status": "OK",
      "Inverters": [
        {
          "SerialNumber": "RB20061402xxxxxx",
          "PowerMode": {
            "InverterMode": "Auto",
            "PowerW": 0
          }
        }
      ]
    },
    {
      "TimestampUtc": "2024-05-27T04:50:26Z",
      "SiteId": "S4d2f2axxxxx",
      "Phases": [
        {
          "Id": "A",
          "ActiveExportedPowerInstantaneouskW": 0,
          "ActiveImportedPowerInstantaneouskW": 5.109,
          "VoltageInstantaneousV": 234.4,
          "CurrentInstantaneousA": -22.63,
          "PowerFactorInstantaneousMinus1to1": -0.969
        },
        {
          "Id": "B",
          "ActiveExportedPowerInstantaneouskW": 0,
          "ActiveImportedPowerInstantaneouskW": 0,
          "VoltageInstantaneousV": 7.7,
          "CurrentInstantaneousA": 0.08,
          "PowerFactorInstantaneousMinus1to1": -0.167
        },
        {
          "Id": "C",
          "ActiveExportedPowerInstantaneouskW": 0,
          "ActiveImportedPowerInstantaneouskW": 0,
          "VoltageInstantaneousV": 7.1,
          "CurrentInstantaneousA": 0.08,
          "PowerFactorInstantaneousMinus1to1": 0.999
        }
      ],
      "FrequencyInstantaneousHz": 49.97,
      "BatterySoCInstantaneous0to1": 0.98,
      "PvPowerInstantaneouskW": 0.284,
      "InverterTemperatureC": 34.2,
      "BatteryPowerNegativeIsChargingkW": -1.407,
      "PvAllTimeEnergykWh": 22523.9,
      "ExportAllTimeEnergykWh": 20247.28,
      "ImportAllTimeEnergykWh": 17083.46,
      "LoadAllTimeEnergykWh": 35065.4,
      "Status": "OK",
      "Inverters": [
        {
          "SerialNumber": "RB20061402xxxxxx",
          "PowerMode": {
            "InverterMode": "ChargeBattery",
            "PowerW": 2163
          }
        }
      ]
    }
  ]
}

cabberley, 2024-05-31:
@homeskool2 thanks for providing the data sets.
I'm working on a solution to o multiple sites and multiple inverters atm.
I will update this thread when I get something that can be tested. 
Shouldn't be too far away if I find the time over the weekend,

homeskool2, 2024-05-31:
Awesome thanks

cabberley, 2024-06-03:
So I have completed the first part of building a new multi inverter Integration and it's ready for some feedback.

https://github.com/cabberley/HA_RedbackTech

This uses a different custom components folder to Juice's Integration so you can run them side by side. 

It will create one configuration per API credentials and will create 2 devices 1 is the Inverter/grid and the other is the Battery per RedBack Hybrid Inverter. 
It should create 2 more devices for each inverter you have and surface all the data for each system. 

This version also gives you detailed data about each Battery, Cabinet and MPPT string.

It doesn't have icons/logos added to HA brands yet so the redback logos don't show up yet for it. 

The next version I will add in inverter control.

juicejuice, 2024-06-03:
Thanks @cabberley this looks really cool. I like how you've separated the Redback connection code into a Python package. I'll try and test this out when I get chance. No point adding multiple inverters to my Redback integration if you've already got there. The ability to control the inverter will be pretty neat too!

cabberley, 2024-06-03:
Thanks @juicejuice it has been a challenge unravelling how HA actually does things. But now that I understand it a lot more, I'm pretty impressed by what you can do with it. 

With the way I built out the new redback library to talk to the API, extending it to control the inverter should be fairly easy now. I think I won't need the portal and can use the schedules. I will hide all the complexities of what goes on in the backend with them.

homeskool2, 2024-06-03:
thanks guys i actually have some maintenance being run tomorrow do to the stats being out of whack so ill try after that's sorted.

it does connect and pick up 2 devices

cabberley, 2024-06-07:
Just released a new version with full inverter control! 2024.6.1
Make sure you read the readme to understand the inverter control!
HTTPS://github.com/cabberley/ha_redbacktech
Enjoy and let me know how goes !!!

homeskool2, 2024-06-29:
hey sorry been away on holiday

all seems to be working great!!

closed 2024-10-09 by juicejuice