juicejuice/homeassistant_redback #27: Credential Expiration

gth-myy-name, 2024-06-17:
Hi Juice

I suspect this is a Redback issue rather than an integration thing - but just in case I'm wrong...

My entire Redback integration stopped today (everything is unavailable), with an "Integration Requires Reconfiguration" error in the Notifications. Went to put the credentials back into the system - copy/pasted from the file I kept of the Swagger system that Redback gave me access through - but "Something Went Wrong".

On closer checking of the original credential file, it indicated an expiry date in April 2024. I think I was one of the first to be using the integration - so perhaps may also be one of the first to have credentials expire as well? :( Has this happened to anyone else?

I have requested a new set of credentials from Redback - will have to wait and see what response I get...

Cheers,

Pcal

GrantKeymer, 2024-06-17:
I was just about to write a separate post concerning Reconfiguration, so instead of doing that, I'll add my own observations here:

1) Late last year, my credentials expired, as @Pcal has found.

2) Redback issued new credentials pretty quickly, so no worries there.

3) After saving in a file, I entered them and within a minute or two, the Redback integration came back online.

4) Since that time, I've needed to re-enter the credentials at least twice.  It happened again last night.  Prior to receiving new credentials, I never needed to re-enter them.

@juicejuice - Just wondering if there has been a change of policy by Redback regarding credentials, or is there some mechanism within your integration that periodically requires credentials to be re-entered?

cabberley, 2024-06-17:
@juicejuice in your code you set the bearer token refresh to now+ expires in seconds. Edge case can be that the token has expired if a call is made in those last seconds as by the time now+expires is calculated it is actual beyond the token life. 

Historically we would make an attempt at 50% of life to renew it. If the token has expired you will likely get a http 500 error.
I had a big in my code o found earlier today, adjusted my expiration and renew to occur if it is within the last 5 minutes of the 60 minute life that the Redback API is setting.

juicejuice, 2024-06-27:
Thanks @cabberley, good point, I would prefer to use a standard OAuth2 library - HA includes this - but I just threw my code together quickly. I think your new integration design around multiple inverters is a better approach.
Anyway, I have now added a 5 min offset, as per your suggestion, to account for the edge case you have highlighted. The retry loop was the previous way that this edge case got resolved but no point triggering a retry if it can be avoided.
@gth-myy-name did you manage to get your issue resolved now?

cabberley, 2024-06-27:
@juicejuice here is the real API endpoint btw. Only got this a few days ago.
[image]

sethkor, 2024-07-19:
This just happened to me I think.  I was prompted to reenter my credentials again in HA and it came back straight away.

GrantKeymer, 2024-07-19:
Yes, this just happened again to me as well.  Almost exactly a month since the last time.

gareth-shakewell, 2024-07-19:
So weird - my API credentials have never expired!

juicejuice, 2024-07-19:
So weird! My API credentials have never expired...

sethkor, 2024-07-19:
first time for me.  I didn't request new ones but maybe the integration needs to get an new token rather than prompt user to reenter?

sethkor, 2024-07-21:
It happened again last night

juicejuice, 2024-07-22:
Yes, I jinxed myself because my credentials have now expired on both July 19 and 21! I've been running the same credentials since 2022 and haven't changed any of that code in the time, so Redback must be doing something on their servers? Maybe they had to reboot/reset/restore due to Crowdstrike issues??

cabberley, 2024-07-22:
Curious to know if you restart HA if it reconnects, or you have to reset/update your secretid and secret?

cabberley, 2024-07-22:
> Yes, I jinxed myself because my credentials have now expired on both July 19 and 21! I've been running the same credentials since 2022 and haven't changed any of that code in the time, so Redback [...]

Pretty sure their backend systems weren't affected by Crowdstuck.

An OAuth Authentication has 2 parts, the secret ID and Secret and then a short lived token for subsequent communications. 
OAuth Process:
1. New session establishment requires requester to send Secret ID and Secret to Server
2. These are validated and if valid a Bearer Token is issued amongst its contents is a Valid Start and End Datetime. Redback Tokens are set with a 3600 second Lifetime, from memory. 
3. If before the End DateTime is reached the client has not negotiated a refreshed token it will behave as if it is not a valid user and treat the request as an unauthenticated session.
4. Programmatically, traditionally we would request a new token at 50% of the expiry time (3600 seconds expiry, try for new token after 1800 seconds) if that fails you still have the original valid token to continue to use with no pressure for getting a new one urgently, and then try again in another 50% of remaining time (1800 + 900 = 2700 seconds after token issued)

From some of the comments above that after re-entry of your originally Secret ID and Secret, it started to work again, I would suspect the token refresh failed.

Hope that helps

juicejuice, 2024-07-22:
> From some of the comments above that after re-entry of your originally Secret ID and Secret, it started to work again, I would suspect the token refresh failed.

If it happens again I'll examine the debug logs. The HA log just reports:
`2024-07-21 22:53:37.204 ERROR (MainThread) [custom_components.redback] Authentication failed while fetching redback data: Invalid credentials`

It would be more informative to retrieve the Redback API error that is triggering the reauth (e.g., [line 55 of config_flow.py](https://github.com/juicejuice/homeassistant_redback/blob/17ec976d1b5c9f55e1e1094f902c816d5ad7060c/custom_components/redback/config_flow.py#L55)). It's weird that it started happening all of a sudden.

cabberley, 2024-07-22:
> If it happens again I'll examine the debug logs. The HA log just reports:
>
> `2024-07-21 22:53:37.204 ERROR (MainThread) [custom_components.redback] Authentication failed while fetching redback data: [...]

I think you will find that it doesn't matter if the token is expired or secret is wrong, it will be the same HTTP error code, I suspect the RestAPI doesn't discriminate why you request authorisation is not valid. 

If you want to debug it you probably want to get the bearer token expiry datetime. Or the daytime of issuance and lifetime values. See how that aligns with the current time.

gth-myy-name, 2024-10-05:
HI Juice,

Sorry haven't been back here for a while. Set new credentials and all worked fine for a while. But more recently, the credentials are expiring ever more frequently. Logs show it has gone down 3 times in the last 16 days. Never been any issue putting credentials back in again, just become a pain to have to keep doing it...

gth-myy-name, 2024-10-05:
Correction, Make that 4 times in 16 days. The last reconfiguration only lasted 3 hours...

GrantKeymer, 2024-10-05:
I've also had to reconfigure twice in recent days.  Will be interesting to see if the reconfiguration I've just done lasts longer than 3 hours.

juicejuice, 2024-10-06:
Yes mine has also expired 4 times over the last week. It says "invalid credentials" but then I use the same credentials as always and they work. Maybe it needs some retry logic before it gives up and declares invalid credentials, or there might be a subtlety to the error condition that needs to be accounted for.

oz-glenn, 2024-10-07:
Just chiming in to say my token “expired” 3 times yesterday.

juicejuice, 2024-10-08:
Yes it is really annoying, mine just did it again. I've changed jobs and laptops lately so I'm setting up a new dev environment so I can make some changes to fix this.

juicejuice, 2024-10-08:
@oz-glenn, @gth-myy-name, @GrantKeymer I can see a few potential causes in the code based on errors from Redback API. If you are are able, please enable debug logging for Redback integration (from the device page it has the option). We can then catch the error red-handed and I can fix it. Unfortunately, it will spam your HA logs with lots of Redback messages as the debug log records every sensor update, every minute... :(

oz-glenn, 2024-10-08:
OK, I've enabled it. Let's wait and see.

GrantKeymer, 2024-10-08:
I've enabled Debug Logging.  My full log goes back to June, so there will be a huge number of entries to search through if anything happens.  Maybe I should restart HA?

juicejuice, 2024-10-08:
> I've enabled Debug Logging. My full log goes back to June, so there will be a huge number of entries to search through if anything happens. Maybe I should restart HA?

Your call, I would be happy to process the logs for you but I am concerned they may contain some form of personal data. If you have a tool like Notepad++ you should be able to search for "invalid credentials" from the Redback integration, then see the API error in one of the lines above.

gth-myy-name, 2024-10-09:
Hi Juice - it just went down again. Log doesn't say a whole lot...

```
This error originated from a custom integration.

Logger: custom_components.redback
Source: helpers/update_coordinator.py:407
integration: Redback Technologies Portal (documentation, issues)
First occurred: 8:59:59 PM (1 occurrences)
Last logged: 8:59:59 PM

Authentication failed while fetching redback data: Invalid credentials
```

Hope it's useful

oz-glenn, 2024-10-09:
Same. Right before dinner here so I was going to have another look after dinner. That’s all I found on first glance.

Despite the integration being red ringed and labeled with “debug logging enabled” I didn’t see any debug style logging around the error.

juicejuice, 2024-10-09:
I think we all must have had this just this evening. It happened approximately 8.30pm QLD time here.
You need to click "Download full log" from the Settings -> System -> Logs panel. You then have to unzip and look at the log file and look at the bottom for the Redback errors.
Mine says:
`2024-10-09 20:30:03.448 DEBUG (MainThread) [custom_components.redback] API error: 500 Internal Server Error. <html><head><title>500 - The request timed out.</title></head><body>  <font color ="#aa0000">         <h2>500 - The request timed out.</h2></font>  The web server failed to respond within the specified time.</body></html>`
Can you guys confirm if this is the same error message on your side?

gth-myy-name, 2024-10-09:
I'm combing through the main log now (editing out everything that doesn't say redback somewhere).  Will do a quick search for your error message...

juicejuice, 2024-10-09:
Ok the key phrase is "API error"

juicejuice, 2024-10-09:
> 2024-10-09 20:59:59.283 DEBUG (MainThread) [custom_components.redback] API error: 500 Internal Server Error. <html><head><title>500 - The request timed out.</title></head><body>  <font color ="#aa0000">         <h2>500 - The request timed out.</h2></font>  The web server failed to respond within the specified time.</body></html>
> Exactly 1.5 hours difference in the timestamp - which if I'm not mistaken
> corresponds with the gap between Qld and SA at the moment...

Perfect. So the Redback server is undergoing some intermittent error
condition which is causing the integration to lose connection. I'll look
into a fix tomorrow. Thanks heaps for the help tracking this down!

oz-glenn, 2024-10-09:
`2024-10-09 18:30:09.861 DEBUG (MainThread) [custom_components.redback] API error: 500 Internal Server Error.  500 - The request timed out. The web server failed to respond within the specified time.

2024-10-09 18:30:09.862 ERROR (MainThread) [custom_components.redback] Authentication failed while fetching redback data: Invalid credentials

2024-10-09 18:30:09.862 DEBUG (MainThread) [custom_components.redback] Finished fetching redback data in 230.317 seconds (success: False)`

Same error, with a couple afterwards. Nothing before.  Times line up. I'm in W.A. Also the fact we all had it at the same time suggests the problem is caused by Redback, which doesn't surprise me. (I hit 1 year installed 5 days ago and it's only been fully working 47% of the time).

cabberley, 2024-10-09:
I suspect there is actually no error, except for the API endpoint being overloaded. We saw a similar problem with Solcast.
I suspect it is caused by everybody's HA time being in sync, the HA coordinater effectively has everyone hitting the Redback API at exactly the same time and overloading it.
What @juicejuice and my integration needs to do is add a randomizer to the data coordinator to randomise the time the integration polls Redback. Pick a random number between 0&10 and delay the request randomly between 1 and 10 seconds.

GrantKeymer, 2024-10-09:
I'm seeing these in the log every so often, but they look to be just a warning and may not be anything to do with the Redback integration.  The keyword API is present, so my search found this line...

2024-10-09 23:07:55.914 WARNING (MainThread) [homeassistant.components.websocket_api.http.connection] [547414261184] Disconnected: Did not receive auth message within 10 seconds

This is what happened last night when my system disconnected...

2024-10-09 23:29:42.667 ERROR (MainThread) [custom_components.redback] Authentication failed while fetching redback data: Invalid credentials

It seems I'm not getting as much detail as the rest of you.

juicejuice, 2024-10-09:
> I suspect there is actually no error, except for the API endpoint being overloaded. We saw a similar problem with Solcast. I suspect it is caused by everybody's HA time being in sync, the HA [...]

That's a good idea, I'll look into that too, thanks!

juicejuice, 2024-10-09:
OK I've just committed a fix that avoids re-authentication when "500 Internal Server Error" is returned by the Redback API service. I'm going to update my HA now and keep an eye out for any further errors. We can probably disable debug logging again unless the problem reoccurs. Try it out and let me know?

gth-myy-name, 2024-10-14:
Sorry Juice - Just gone down again - 9:04:34 (SA Summer time GMT+10.5).

Re-Authenticated immediately (manually) without issue.

oz-glenn, 2024-10-14:
Came here to say this same thing.

juicejuice, 2024-10-14:
Yes same here too. It is definitely some sort of intermittent problem on their server. Due to the intermittent nature I cannot reproduce it upon demand, meaning I can't test my fixes straight away. I've enabled debug logging again and will try catch it next time it happens and see what is going on this time. Perhaps it's a different error code?

gth-myy-name, 2024-10-14:
Given that it seems to work right away on a re-authentication, is it possible to test for the authentication failure, and when detected just try again random(X) seconds later using the existing credentials. Still go to the manual Re-Authentication process if it fails Y times in a row (2 or 3?)...

Just thinking what I would do if I knew how.

juicejuice, 2024-10-14:
> Given that it seems to work right away on a re-authentication, is it possible to test for the authentication failure, and when detected just try again random(X) seconds later using the existing [...]

Yes exactly, the problem is that we need to know if it's a real authentication issue or fake issue. For fake issues it will just wait and retry. For real issues we need to trigger a re-authentication. The problem is that it doesn't seem to detect the fake issues, that's what I'm working on. It is really hard to fix without being able to reproduce the error condition.

cabberley, 2024-10-17:
> That's a good idea, I'll look into that too, thanks!

@juicejuice i added a randomised variable to my HA coordinator, but that doesn't stop the intermittent timeouts.

juicejuice, 2024-10-17:
> @juicejuice i added a randomised variable to my HA coordinator, but that doesn't stop the intermittent timeouts.

Interesting... are you trapping the intermittent error condition and timeout? Is it showing up as 500 Internal Server Error in your logs?

GrantKeymer, 2024-10-17:
Another disconnection occurred 36 minutes ago (about 0110 UTC).

oz-glenn, 2024-10-17:
> Another disconnection occurred 36 minutes ago (about 0110 UTC).

Not for me.

juicejuice, 2024-10-17:
> Another disconnection occurred 36 minutes ago (about 0110 UTC).

Not for me either. Interesting!

gth-myy-name, 2024-10-17:
I've had 2 disconnections today. One about 10 minutes after midnight which I didn't see for a number of hours, and again around the time that @GrantKeymer reported. I switched debug logging back on after that... Curiously, the second event happened only moments before Chris posted about his randomised variable. I didn't look at the time, but assumed someone else had reported the outage I had just had and so kept on working.

gth-myy-name, 2024-10-17:
Now I've had 3.

Here is the interesting bit from the log. Hope it is helpful...

```
2024-10-17 18:34:54.791 DEBUG (MainThread) [custom_components.redback] API error: 502 Bad Gateway. <!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
<head>
<meta http-equiv="Content-Type" content="text/html; charset=iso-8859-1"/>
<title>502 - Web server received an invalid response while acting as a gateway or proxy server.</title>
<style type="text/css">
<!--
body{margin:0;font-size:.7em;font-family:Verdana, Arial, Helvetica, sans-serif;background:#EEEEEE;}
fieldset{padding:0 15px 10px 15px;} 
h1{font-size:2.4em;margin:0;color:#FFF;}
h2{font-size:1.7em;margin:0;color:#CC0000;} 
h3{font-size:1.2em;margin:10px 0 0 0;color:#000000;} 
#header{width:96%;margin:0 0 0 0;padding:6px 2% 6px 2%;font-family:"trebuchet MS", Verdana, sans-serif;color:#FFF;
background-color:#555555;}
#content{margin:0 0 0 2%;position:relative;}
.content-container{background:#FFF;width:96%;margin-top:8px;padding:10px;position:relative;}
-->
</style>
</head>
<body>
<div id="header"><h1>Server Error</h1></div>
<div id="content">
 <div class="content-container"><fieldset>
  <h2>502 - Web server received an invalid response while acting as a gateway or proxy server.</h2>
  <h3>There is a problem with the page you are looking for, and it cannot be displayed. When the Web server (while acting as a gateway or proxy) contacted the upstream content server, it received an invalid response from the content server.</h3>
 </fieldset></div>
</div>
</body>
</html>

2024-10-17 18:34:54.791 ERROR (MainThread) [custom_components.redback] Authentication failed while fetching redback data: Invalid credentials
2024-10-17 18:34:54.793 DEBUG (MainThread) [custom_components.redback] Finished fetching redback data in 177.672 seconds (success: False)
```

juicejuice, 2024-10-17:
Thanks! Same here, it is a **502 Bad Gateway** error. I'm thinking if we treat all the 500 series errors as not authentication errors then it should avoid the issue. I'll make a patch shortly.

juicejuice, 2024-10-17:
OK guys I've pushed another update the traps all 500-599 errors to ensure they do not trigger a re-authentication. I know this has been quite frustrating. Please do let me know if you find any other error cases!

samuelyeats, 2024-10-18:
Hey there - I've got new API credentials that work directly on the Redback API. I'm trying to add my Redback entity, but I get an "Unexpected error" when clicking save on the Add entity, and a 500 error on the Home Screen:

Failed setup, will retry: HTTP error: 500 Internal Server Error. {"Error":"Unknown error, please report to Redback Technologies"}

samuelyeats, 2024-10-18:
2024-10-18 20:12:33.756 DEBUG (MainThread) [custom_components.redback] Syncing data with Redback (entry_id=01JAFEA8X7X1RFHSE6ZH7PRPSJ)
2024-10-18 20:12:34.220 DEBUG (MainThread) [custom_components.redback] Finished fetching redback data in 0.463 seconds (success: False)
2024-10-18 20:12:44.411 DEBUG (MainThread) [custom_components.redback] Syncing data with Redback (entry_id=01JAFEA8X7X1RFHSE6ZH7PRPSJ)
2024-10-18 20:12:46.724 DEBUG (MainThread) [custom_components.redback] Finished fetching redback data in 2.313 seconds (success: False)

juicejuice, 2024-10-20:
Hi @samuelyeats that sounds like a different issue from this one. The 500 "unknown error" doesn't sound very promising though does it? If the issue exists, please open up a new ticket and I can see if I have any ideas to help you.

gth-myy-name, 2024-11-03:
Thanks Juice,

Rock solid the last few weeks. Whatever the root cause, I think you've successfully "de-tuned" it. Cheers! :)

closed 2024-11-03 by gth-myy-name

juicejuice, 2024-11-03:
I can confirm my system is humming along again too :) thanks for the feedback and for your help finding the issue!