juicejuice/homeassistant_redback #28: Managing battery dispatch

Mark-Hetherington, 2024-07-03:
This is intended more as a discussion than an issue. I'm exploring controlling the battery dispatch from Home Assistant. Currently I manage another battery system from Home Assistant via modbus, so it would be good to also coordinate the operation of the Redback.

It looks like there are a few options here. 

1. The Public API has the option of [creating a schedule](https://api.redbacktech.com/docs/index.html#/Schedule/Schedule_CreateSchedule). This makes the most sense if a plan is created based on forecast loads or pricing. But it might require rewriting the scheduling if conditions change and the plan is re-evaluated. It could I suppose achieve near real-time control if the schedule is adjusted every few seconds. However, there are enough modes that it should be possible to set the objective and leave the real-time control up to the inverter. I'm not quite sure what some of the modes do though. 
2. The Public API has an option of creating an operating envelope, where some parameters can be set. Again, it's scheduled. But it looks like it might be a good option for curtailing solar if the feed in is predicted to go negative.
3. The private API has the ability to set a schedule, with fewer modes than offered by the public API. At least as observed in a web browser in the web portal. The portal also allows for setting a charge or discharge mode and rate instantaneously, rather than on a schedule. This would fit in better with how I'm controlling the other system where I send a command for power transfer rate and then expect the inverter to start doing that.

I have a slight issue with option 1, where there seems to be a disconnect between the web portal and the API. If I create schedules in the API, they don't appear in the web portal (and I'm not sure they work). Similarly if I create a schedule in the web portal, I don't see it in the API (but they definitely do work).

I'm not sure how to map this in terms of the Home Assistant UI. Setting schedules with modes and other parameters is different to a set of simple independent control like a slider or a switch. The start/finish, mode and power level are all interdependent.  Perhaps it makes more sense to link this to something like [EMHASS](https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126) which already operates on the concept of a forward plan or schedule. 

I don't know if anyone else has experimented with any of this? Or if anyone has use cases that would be good to consider?

cabberley, 2024-07-03:
@Mark-Hetherington have a look at my integration, enables you to do all of that and more with the Redback
https://github.com/cabberley/HA_RedbackTech.
Setting a schedule through the portal or API sets the data in the same place within Redbacks backend, so you should see it in both places. My testing it does show up.
Looking through your other issue with your Redback not registered properly, I would suggest trying support again, now that they have new owners and funding things are looking up. If you don't have any luck let me know and I can try and help you get it sorted.

Mark-Hetherington, 2024-07-03:
Oh, wow, thanks for letting me know about that integration. I'll take a look.
I think there is something a bit funky with my inverter registration, as it was replaced after a lightning strike. But I might report the mismatch in the scheduling along with following up the other issues.

cabberley, 2024-07-03:
I suspect that the inverter schedule and envelope controls won't show up until we get your battery count above 0.
I will if I get time create a patch to enable them first your system until you can get the data fixed.
The code in the redbacktechpy python library published separately is where the patch needs to go.

cabberley, 2024-07-03:
@Mark-Hetherington I have made a patch release that should enable all the functionality in my integration for your situation.
If you have installed my integration after it has been installed
To use it, Modify your HASS custom_components/redback_tech/manifest.json file change:
 "redbacktechpy==2024.7.0"
to
 "redbacktechpy==2024.7.1a1"

restart HASS, it will need to download a newer python library so will be a little slower first reload after the change. If you have any issues let me know on my HA Redback repo.

Mark-Hetherington, 2024-07-05:
Thanks very much for your efforts. I hope I'll get some time to tinker with it this weekend.

Mark-Hetherington, 2024-07-17:
I've installed it and run it, unmodified, and was getting data. The scheduling buttons also showed up!

After making the modification to run redbacktechpy==2024.7.1a1 it seems like it's trying to access data that isn't supplied from the API for our system (soc_data['Data']['MinSoC0to1'] is None). I'm not that surprised having seen these inconsistencies previously. 

I am going to revert it, and tinker with the unmodified version, while I follow up with Redback support. I might follow up anything else on your repo.

cabberley, 2024-07-17:
@Mark-Hetherington would you like to email me directly at chris@abberley.com.au with some details about the last support ticket number from Redback and what happened to your inverter? And what is missing in your Redback config. 
I can try to get it fixed for you.

closed 2024-10-09 by juicejuice