juicejuice/homeassistant_redback #30: Unexpected error

samuelyeats, 2024-10-21:
Hey @juicejuice - thanks for looking into this - I've got new API credentials that work fine when tested against the Redback API. 

I'm trying to add my Redback entity on your latest version from last week, but I get (1) an "Unexpected error" when clicking save on the Add entity, and  (2) a 500 error on the Home Screen:
Failed setup, will retry: HTTP error: 500 Internal Server Error. {"Error":"Unknown error, please report to Redback Technologies"}

2024-10-18 20:12:33.756 DEBUG (MainThread) [custom_components.redback] Syncing data with Redback (entry_id=01JAFEA8X7X1RFHSE6ZH7PRPSJ)
2024-10-18 20:12:34.220 DEBUG (MainThread) [custom_components.redback] Finished fetching redback data in 0.463 seconds (success: False)
2024-10-18 20:12:44.411 DEBUG (MainThread) [custom_components.redback] Syncing data with Redback (entry_id=01JAFEA8X7X1RFHSE6ZH7PRPSJ)
2024-10-18 20:12:46.724 DEBUG (MainThread) [custom_components.redback] Finished fetching redback data in 2.313 seconds (success: False)

juicejuice, 2024-10-22:
OK I have something you can try, if you are familiar with command line interface. This should work on Windows 10/11 command prompt or Mac/Linux terminal.

**Step 1. Get auth token from Redback. Use Redback API credentials at XXX and YYY below.**
`curl -X POST -d "client_id=XXX&client_secret=YYY" https://api.redbacktech.com/api/v2/Auth/token`

**Step 2. Step 1 returns an auth token in JSON format, valid for one hour. You need to fish out the really long TOKENSTRING part without its surrounding quotes.**
`{"token_type":"Bearer","expires_in":3599,"ext_expires_in":3599,"access_token":"TOKENSTRING"}`

**Step 3. See if you can access one of the Redback API endpoints, replacing TOKENSTRING with that super long auth token from Step 2.**
`curl -H "authorization: Bearer TOKENSTRING" https://api.redbacktech.com/Api/v2/EnergyData`

If one of the steps doesn't work, you can put a ` -v ` flag before the https://api.redbacktech.com bit to get it to spit out more details. Let us know how you go?

samuelyeats, 2024-10-22:
Hi @juicejuice That seemed to work fine

* Request completely sent off
< HTTP/1.1 200 OK
< Content-Length: 74
< Content-Type: application/json; charset=utf-8
< Date: Tue, 22 Oct 2024 03:17:40 GMT
< Server: Microsoft-IIS/10.0
< Set-Cookie: ARRAffinity=0e5f53c06fe9c8a84e3589fcb7d78a91d5a2ad5b867ea6746e75561303361b04;Path=/;HttpOnly;Secure;Domain=api.redbacktech.com
< Set-Cookie: ARRAffinitySameSite=0e5f53c06fe9c8a84e3589fcb7d78a91d5a2ad5b867ea6746e75561303361b04;Path=/;HttpOnly;SameSite=None;Secure;Domain=api.redbacktech.com
< Request-Context: appId=cid-v1:52b49cfe-5743-4813-b663-63a74f164ccb
< X-Rate-Limit-Limit: 7d
< X-Rate-Limit-Remaining: 19999990
< X-Rate-Limit-Reset: 2024-10-29T03:12:49.3644845Z
< X-Powered-By: ASP.NET
< 
* Connection #0 to host api.redbacktech.com left intact
{"Page":0,"PageSize":100,"PageCount":1,"TotalCount":1,"Data":["S2021893"]}

juicejuice, 2024-10-22:
That's good, thanks! Those are essentially the commands that the Redback integration is using under the covers. However, that still leaves the question of why yours isn't working within Home Assistant. Any chance you can share some screenshots as you step through the setup process so I can see exactly where and how it is going wrong?

samuelyeats, 2024-10-22:
Of course, everything up to date except for todays HA Core

[image]
[image]
[image]
[image]

juicejuice, 2024-10-23:
Thanks - that doesn't look right. I just did a fresh install of the Redback integration here and it looks somewhat different. Is it possible for you to setup from scratch again and record screenshots like mine to compare against? If you can enable debug logging then that might help give some clues as well.

**Step 1 - add integration**
[image]

**Step 2 - enter details**
[image]

**Step 3 - success**
[image]

**Step 4 - check device page**
[image]

**Step 5 - check entities**
[image]

samuelyeats, 2024-10-24:
Thanks for looking into this. This has subsequently been resolved by redback following a firmware upgrade of my system. As a FYI if anyone else experiences this issue, call redback select #3 for upgrade support.

closed 2024-10-24 by samuelyeats

juicejuice, 2024-10-24:
Awesome! Thanks @samuelyeats for reporting back on this.

GrantKeymer, 2025-04-11:
Earlier this week I needed to reconfigure the Redback integration and thought nothing more of it.

However, the same thing happened today and now the reconfiguration is unsuccessful.  I'm see this error message displayed:

`Retrying setup: HTTP error: 500 Internal Server Error. {"Error":"Unknown error, please report to Redback Technologies"}`

Is anyone else seeing something similar?