juicejuice/homeassistant_redback #31: Thanks for the new sensors

GrantKeymer, 2024-12-26:
Hi @juicejuice - Compliments of the season to you, and I want to say thanks for the new sensors you added a few days ago.  It's nice to be able to see how my individual PV Panel Strings are doing, without needing to wade through the Redback Portal.

According to your code, it looks as though there are individual sensors for each battery too, but none have appeared for my SH5000 setup with 4 batteries.  No matter, I'm happy to see the extra PV sensors.

juicejuice, 2024-12-26:
Thanks @GrantKeymer! Seasons greetings in return :)
Yes I did add support for the new EnergyData/Dynamic v2.21 API, seeing as I had a little time spare from my day job. The updated API includes PV string details and individual battery module details. While my code downloads all data I only made sensors for the PV string details and summary battery data, which was mainly what I was interested in. The "missing" sensors would be per-module details of battery current, voltage, power, and state-of-charge. Feel free to add these as a pull request, I couldn't be bothered as I wasn't that interested.
If you are keen the sensor code would go around line 440 of sensor.py. To get the relevant data points for the sensors you probably would want to interpret the Redback API JSON structure around lines 321-323 of redbacklib.py.

GrantKeymer, 2024-12-26:
Thanks @juicejuice - I'll have a poke around as time permits.  As part of the new v2.21 API, have Redback made available the Backup Load power and Non-Backup Load power as individual data values?

Those values are some of the ones I most frequently want to see, and currently, the only way is from the Redback App or Redback Portal.  If they're available through the new API, then it will definitely make sense for me to start work on a Pull Request.

juicejuice, 2024-12-26:
> have Redback made available the Backup Load power and Non-Backup Load power

Sadly not. While my integration code will calculate site load from the available data points there is no way to derive backup vs non-backup load from the public APIs. I do wish they would add that!

BTW, @cabberley's integration uses the "private" API set, authenticated by the regular account username/password. If he is able to access the https://portal.redbacktech.com/charts/GetDataBands API endpoint then that will work for you as that is the same data that drives the portal charting tool.

[image]

cabberley, 2024-12-26:
ATM I only access the portal API/code to set the inverter to Auto mode. I don't poll it for anything. 

There are a few tweaks I have on my backlog for my integration. I can add this to the list, my initial thought would be to make it optional config item. 

@juicejuice and @GrantKeymer are there any other datapoints not available in the API or interest? 

Also a note for those that are looking at individual battery data points. From my observations the only accurate one is the voltage values. The others are all the same and reflect the total values.

The voltage numbers are what is normally used to determine kWh storage for these types of batteries and that's a tricky calculation as it's not a linear relationship. And the stored kWh/voltage is technically calculated after the battery has had a chance to "rest" the voltages during charge/discharge and immediately after stopping need some time first.

Lastaerie, 2024-12-26:
Hi guys,  just wanted to say a massive thankyou for your efforts on this integration over the past year and wish you a great Christmas and a Happy New year.  Cheers

juicejuice, 2024-12-27:
> are there any other datapoints not available in the API or interest?

TBH, now that you identified the PV and Battery details in the v2.21 APIs, the only important missing data points are what @GrantKeymer is asking about - Load and Backup Load. From my perspective, nothing else of particular interest appears to be missing.

GrantKeymer, 2024-12-27:
> @juicejuice and @GrantKeymer are there any other datapoints not available in the API or interest?
>
> Also a note for those that are looking at individual battery data points. From my observations the [...]

@cabberley - Thanks for the warning about the individual battery data points.  It's mainly the 4 SoC readings I'm interested in, as my set of batteries are not all the same, and I may well end up with 2 of them reaching a premature end, as happened 18 months ago.  So I want to keep a close eye on any mismatch.

Apart from the SOCs, the separate readings for AC Load and Backup Load are what I'd really like to be able to access from HA.

closed 2025-03-03 by GrantKeymer