juicejuice/homeassistant_redback #32: TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'

BigByte76, 2025-03-02:
Have been using the integration for several months without any real issue and have just started receiving the below information. Have removed removed the integration and reloaded it but the error remains. Error message first appeared on the 27th of Feb and hasn't worked since.
```

This error originated from a custom integration.

Logger: custom_components.redback
Source: helpers/update_coordinator.py:380
integration: Redback Technologies Portal (documentation, issues)
First occurred: 5:12:56 PM (7 occurrences)
Last logged: 5:17:23 PM

Unexpected error fetching redback data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/redback/coordinator.py", line 46, in _async_update_data
    self.energy_data = await self.redback.getEnergyData()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/redback/redbacklib.py", line 318, in getEnergyData
    self._energyData["ActiveExportedPowerInstantaneouskW"] = sum(list(map(lambda x: x["ActiveExportedPowerInstantaneouskW"], self._energyData["Phases"])))
                                                             ~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'
```

juicejuice, 2025-03-02:
Sorry to hear that! The Redback integration is still working on my Home Assistant so I don't have an immediate answer. Is it possible that you can send me the output of one of the API requests from **api.redbacktech.com**? You need to login with your credentials, find your site ID, then use the **/Api/v2/EnergyData/{siteId}/Dynamic** API. If you can send me the results of that API call then I can see what has gone wrong. 
BTW you will receive a JSON text file as the output from that API. It will contain your Site ID and Serial Number which you probably want to clear out from the file before sending it over - I don't want or need any of your personal details :)

BigByte76, 2025-03-02:
Thanks! I tried going through the logs further but I don't have anything showing up for the 27th of Feb to see what changed for the home assistant side of things and no updates where done then as I was away from home at the time.

Here is what the API output is showing with my Site ID / Serial Number.

```
{
  "Data": {
    "TimestampUtc": "2025-03-02T08:52:54Z",
    "SiteId": "***",
    "Phases": [
      {
        "Id": "A",
        "ActiveExportedPowerInstantaneouskW": null,
        "ActiveImportedPowerInstantaneouskW": null,
        "VoltageInstantaneousV": 237.8,
        "CurrentInstantaneousA": null,
        "PowerFactorInstantaneousMinus1to1": null
      }
    ],
    "FrequencyInstantaneousHz": null,
    "BatterySoCInstantaneous0to1": 0.78,
    "PvPowerInstantaneouskW": 0,
    "InverterTemperatureC": 53,
    "BatteryPowerNegativeIsChargingkW": 0.919,
    "PvAllTimeEnergykWh": 22379.69,
    "ExportAllTimeEnergykWh": 14674.45,
    "ImportAllTimeEnergykWh": 3453.488,
    "LoadAllTimeEnergykWh": 9805.12,
    "Status": "OK",
    "Inverters": [
      {
        "SerialNumber": "***",
        "PowerMode": {
          "InverterMode": "Auto",
          "PowerW": 0
        }
      }
    ]
  },
  "Metadata": null
}
```

juicejuice, 2025-03-02:
Perfect, thanks! I've never seen before but the "null" values are what is causing the issue. I'll make some updates here to convert those values to zeros. It is very unusual though because that means you won't get any proper data for your import/export power, current (amps), or frequency (Hz). I would query this with Redback Support to make sure your equipment doesn't have some sort of issue.

juicejuice, 2025-03-02:
That was quick and easy. The latest commit I just made will avoid those errors by converting the null values to zeros. While that fixes the error message it does mean you will just have zeros for all sorts of important data including your power import/export. Does your Redback app and web dashboard still work? I think they all draw data from the same database.

BigByte76, 2025-03-03:
Thanks! That fixed the integration not starting, but as you mentioned, there are other issues with my system. The app shows the grid being disconnected. I went out to the switchboard and saw that the SM3000 coms light is just flashing showing that it is transmitting data but it seems the inverter isn't receiving it.

juicejuice, 2025-03-03:
Sorry to hear that! Not much else I can do then, I'm afraid. I'll close this ticket for now.

closed 2025-03-03 by juicejuice