juicejuice/homeassistant_redback #33: Unexpected Error 500

GrantKeymer, 2025-04-11:
Earlier this week I needed to reconfigure the Redback integration and thought nothing more of it.

However, the same thing happened today and now the reconfiguration is unsuccessful. I'm see this error message displayed:

`Retrying setup: HTTP error: 500 Internal Server Error. {"Error":"Unknown error, please report to Redback Technologies"}
`

Is anyone else seeing something similar?

oz-glenn, 2025-04-11:
I'm getting the same error. Interestingly the redback app was offline for a while. Now it's back, but the component now has this error.

juicejuice, 2025-04-11:
Same here:
`Failed setup, will retry: HTTP error: 500 Internal Server Error. {"Error":"Unknown error, please report to Redback Technologies"}`

juicejuice, 2025-04-11:
This error actually doesn't come from the integration. I've tried direct API calls from a Python script and it returns the same 500 error. It looks like Redback's own API servers are experiencing a fault. The only option would be to file a support ticket with Redback :(

cabberley, 2025-04-11:
Something was going on earlier today.
These are from my integration, it does seem to have recovered now tho.

[image]
 [image]

oz-glenn, 2025-04-11:
Mine hasn’t recovered yet

cabberley, 2025-04-11:
@juicejuice do you log failed authentications?

oz-glenn, 2025-04-11:
I've been mucking around with https://api.redbacktech.com/docs/index.html and on there I get a 500 error for /Api/v2/EnergyData/Static/BySerialNumber/{serialNumber} but I get a result for /Api/v2/EnergyData/Dynamic/BySerialNumber/{serialNumber}. 

Something weird going on at their end.

juicejuice, 2025-04-11:
I can confirm the same here. It is the "Static" API method that is failing with a 500 error. The other API methods seem fine. I hope Redback haven't lost their commissioning database! I'll file a support ticket otherwise will need to see if I can simply "blank out" the missing Static data in the case that API method call fails. The Dynamic data is mostly what we need from the inverter anyway.

cabberley, 2025-04-11:
@juicejuice try using this endpoint for your static data instead.

Api/v2/EnergyData/{self.siteId}/Static

This is the one I use and it's still working.

juicejuice, 2025-04-11:
Thanks @cabberley but that is actually the one I'm using too :( I tried the other "BySerialNumber" method and that fails the same way. This falure is apparent even using Redback's own Swagger API test site. It returns an HTTP 500, which usually indicates a server error.

GrantKeymer, 2025-04-11:
A lot of data is missing from the Redback app.  All the current weekly charts have been truncated to low levels and previous weekly/monthly data has vanished.  Seems like a major meltdown has happened at their end, and of course it had to be late on Friday afternoon.  Murphy strikes again 😒

LachyGoshi, 2025-04-12:
Yeah all my history has disappeared in the app and portal also. I've been trying to get my API key renewed for weeks, the portal is now meant to let you renew it, however I type in a name for it and click the OK button, it spins for 10 seconds, changes colour but nothing happens and no new key is generated. When contacting support via both the portal and email, when I did finally get a reply after a few weeks, in typical crapback support, they advised me to use the portal to generate a token. Another review and experience of absolutely useless support as always from redback technology.

I haven't logged a support for my missing historical data yet (had been away so only checked last night) and noticed, seems like I'm not the only one missing the data.

juicejuice, 2025-04-12:
Yes my historical data on the portal is all gone too. I noticed the Redback CEO, Adrian Knack, moved on a few weeks back. Coincidence?

gth-myy-name, 2025-04-12:
Nothing new to add beyond what you have all noted already, except that mine went down twice in the week before yesterday's meltdown. Once for a few minutes, and the second time for a couple of hours. Both times it righted itself before I had a chance to look at at - so I was hoping it may have done the same again - but clearly not.

Has anyone had any joy playing with direct communication with the inverter itself rather than letting the less than reliable middle man hold everything up?

cabberley, 2025-04-12:
There is no direct access to the redback system at this time.

LachyGoshi, 2025-04-12:
I believe the main OUIJA BOARD is an rpi. They will never give the login details, someone smarter than me may be able to hack it. My guess would be it's just running a standard Linux OS and would have various IO and python scripts running but I got nothing to base that on....

My other idea is to try and tap in to any IO as it's prob all modbus or similar. I think I remember reading years ago the inverters were re branded, so perhaps you could remove the ouija board all together and replace with your own custom pi haha, how goot would that be!

I already monitor the battery stack using an esp on the pylontech modbus port. Just read only because I don't want to stuff anything up from the redback side of things.

Anyway, just refreshed the portal and looks like the history is back! Now I just need to hope I can finally renew my API key....

GrantKeymer, 2025-04-12:
Yes, I can also see the history again, both in the portal and in the Redback app.

However, the '500 Internal Server Error' message persists, even though I reloaded the HA integration.

juicejuice, 2025-04-12:
Yes my history is back now too but the HTTP 500 error remains. I'm away this weekend so no chance to patch up the integration until next week. I do wish we had direct access to the Raspberry Pi Ouija board. It is locked down pretty tight last time I looked at it, although I don't think they patch it often so there are probably vulnerabilities available these days.

juicejuice, 2025-04-14:
Hi team, is everyone's integration working again now? Mine just fixed itself today too. It wasn't working first thing this morning but fixed itself around 11.24am. I found a workaround (avoiding the use of Static data in the integration) but won't make any changes unless someone is still experiencing the issue.

GrantKeymer, 2025-04-14:
It's all good for me, and has been for the last 3 hours or so.  Many thanks for checking 👍

closed 2025-04-15 by juicejuice

juicejuice, 2025-04-17:
Hey all, I got an update from Redback support that explains the outages.

> A server migration was completed last Friday, 11/04/25.
> However, the IT team has informed us that the merging of user data is still in progress.
> We apologize for the inconvenience this has caused, and the issue should be resolved soon.

cabberley, 2025-04-17:
@juicejuice sent you a linked in message