juicejuice/homeassistant_redback #34: Sensors locked on false value?

gth-myy-name, 2025-04-16:
Hi Juice,

Everything was working fine for a day or two, but as of just after 4pm this afternoon _all_ my Redback sensors have jumped to impossible numbers and stuck there unmoving since.

For example...

[image]

I appear to be generating 3.4kW in the pitch black.

All sensors jumped to unbelievable numbers at the same time. In the Redback Portal, the correct information is showing so it can't be a hardware failure or coms issue between me and them. There is nothing at all in my HA logs at this time.

Does anyone else have this issue?

gth-myy-name, 2025-04-16:
Just came good by itself...

[image]

More funky stuff happening on the Redback server...

oz-glenn, 2025-04-16:
Mine was out for a similar time but it's also back now. Redback api is not having a good time lately...doesn't bode well for the long term - reinforcing that local control is the best solution.

closed 2025-04-17 by juicejuice