juicejuice/homeassistant_redback #35: Error fetching redback data: HTTP error: 500 Internal Server Error

GrantKeymer, 2025-06-05:
Sadly, this error reappeared at about 8am NZT this morning and persists as of now.  Full text:

'Error fetching redback data: HTTP error: 500 Internal Server Error. {"Error":"Unknown error, please report to Redback Technologies"}'

Is anyone else seeing this?

I've also been locked out of the Redback App for some reason, but not the Redback Portal.

cabberley, 2025-06-06:
Yep the Redback cloud has shat itself. Web portal and mobile app are now dead as well. They were still working earlier and reporting your inverter is offline, but still working. 
I have access to some data directly from my redback which shows at least it is working and charging the battery etc.

cabberley, 2025-06-06:
There is a direct connection that conforms to sunspec. However!! 
1. It's not a normally available option to the Consumer.
2. If you are granted write access to the SunSpec Chanel if it misses a heartbeat from your client it will send the Inverter into shutdown mode. 
3. Access can be read only or read/write.

cabberley, 2025-06-06:
Whinge loud enough and things magically just started working again

juicejuice, 2025-06-06:
@cabberley they must be monitoring the thread :D

cabberley, 2025-06-06:
Pretty sure they aren't 
Pretty sure I know the root cause
Pretty Pi**ed off that I could export during the morning fit peak as they have been better than the evening peaks lately.
Pretty sure my next inverter battery won't be a RedBack

gth-myy-name, 2025-06-07:
Hi guys - is it still down everywhere else?

I'm no longer getting "unavailable" errors - but my battery is listed in the App as being "Disconnected" - and is read in HA as 0%.

While I am very pleased that we are finally getting some rain - it means I don't have enough solar generation to be putting anything into my batteries anyway, so I can't tell if the hardware is actually still functional...

Also noted that the Contact page  in the App has been altered to push me to the Installer (Redback don't want to talk to us any more?). Unfortunately my Installer gave Redback the flick some time ago for poor support -so I can't really go there either if anything is genuinely fritzed.

GrantKeymer, 2025-06-07:
All the sensors in HA are working again, and access to the App has been restored.  I still don't have any charts in the Portal though.

gth-myy-name, 2025-06-07:
OK. Glad its working for you - but according to the App, it seems I don't even _have_ a battery. (Dashboard -> Settings -> Battery Capacity : 0Wh) :(

Have to wait until after the Long Weekend then see who I can contact.

cabberley, 2025-06-07:
So the inside gos for everyone here... after redback was bought/refinanced out of insolvency. The new owning entity has decimated the local Brisbane HQ, there is no technical R&D team in Australia. All the offshore people are new and don't have a full understanding of how the backend cloud works. Certain parts and probably more to follow have been shifted from Azure to Alibaba cloud. No idea what region atm. The two big outages you have all experienced. The first I know from inside info was the num nuts that had no idea what they were doing when they moved some backend stuff to Alibaba. The one the last few days, I suspect they moved more and again had no idea what they were doing. 
While the current and about to be released RedBack models were developed in Brisbane, after that it won't be Australian designed. There is no local technical people any more. 

Personally, I will not be buying anything more from RedBack.

GrantKeymer, 2025-06-07:
Many thanks @cabberley for the 'inside info'.  My SH5000 is on its last legs now, with PV production dropping out almost completely when clouds come over.  From 2, 3, 4kW or more, it suddenly drops to about 200 Watts, before recovering as time goes on.  One cloudy day recently resulted in only 0.96kW being produced!  It was a cloudy day, but not that cloudy.  Typically, I'm seeing 15 to 25kWh on recent days, with the occasional day over 30kWh.  So there was no excuse for the output to drop to less than 1kWh.

Redback have acknowledged this as a known fault, with a number of early-model SH5000s being affected.  Our distributor in NZ (Zen Energy) has stood right behind me all the way, pressing for action from Redback.  As a result, I have a later model SH5000 being sent over by sea freight, due to arrive in the next week or so.  It's this level of support which will keep me coming back to Redback and Zen for upgrades as the years progress.

I hear what you're saying, but won't Redback now be in the same boat as all other suppliers selling equipment which is not designed in Australia?

gth-myy-name, 2025-06-07:
Thanks @cabberley. That's the worst news I've heard in a long time (excluding international politics and all). The ONLY reason I paid _more_ for a _smaller_ Redback system was because their server and tech support was in Australia. It was one of my primary considerations. I wasn't keen to be connected to a European server, but would never have even considered a Chinese one.

Very "Not Happy".

cabberley, 2025-06-07:
So my inside contacts at redback have all left. So beyond what I have divulged above... what happens next I won't have much info about unfortunately.