juicejuice/homeassistant_redback #36: All batteries disconnected from SH5000

GrantKeymer, 2025-06-15:
Yesterday morning, all of my batteries disconnected from the SH5000, just as the sun was starting to shine on the panels.  It looked to me as though it was a hardware fault, but I've just been informed by our installer that this has been caused by a "software glitch on the backend portal".  More tinkering by Redback staff who don't know what they're doing maybe?

Is anyone else affected by this?

juicejuice, 2025-06-15:
@GrantKeymer all looking good here this morning...

GrantKeymer, 2025-07-07:
Bugger!  My batteries disconnected again last night and have stayed that way for 11 hours now.

Can't wait for the new equipment to arrive.

GrantKeymer, 2025-07-20:
@gth-myy-name - Sure, you're welcome to have details of the Battery Reset procedure, as per page 49 of the Installation Manual, where it is described really well.

The On/Off switch referred to is a small rocker switch at the back/left corner of each battery module.  Take note of which battery module is the Master, according to where the cable from the Fan Controller is plugged in.  It has a tag around it labelled 'Inverter Comms' or something like that.

[image]

GrantKeymer, 2025-10-17:
The second and third screens list all the individual cell voltages and temperatures, but the info in the main panel at the top is all the same.  If anyone is interested, I can post details of the hardware I've used to access the Console Port of the Master Battery (first in the chain, starting from the Fan Controller board).

juicejuice, 2025-10-18:
@GrantKeymer very cool! I for one am interested in how you have achieved this.