juicejuice/homeassistant_redback #39: How do I get to enter the Site ID ?

GrantKeymer, 2025-08-04:
This afternoon, I have had a brand new SH5000 inverter installed, and of course the serial number is different.

Using the 'Get available sites' API command, I can now see this:

{
  "Page": 0,
  "PageSize": 100,
  "PageCount": 2,
  "TotalCount": 2,
  "Data": [
    "S17*************",
    "S56*************"
  ]
}

So I have the Site IDs but can't find any way to enter the new one which starts with S56.  In the file Readme.md it says:

`The "Redback Site" field is only required when you have multiple Redback inverters. If you have a single inverter then leave this field alone. For multiple inverters, each inverter is a "site", and you simply indicate which inverter you are setting up (first, second, third, etc.)`

But in spite of editing the `core.config_entries` file in `.storage` to force a re-config and then rebooting HA, I don't see any prompt for 'Redback Site".

What am I doing wrong?

juicejuice, 2025-08-04:
I don't have my test system handy at the moment but you should be able to work it like this. Assuming you had a fresh HA install with no Redbacks added:
1. Install HACS, then install the Redback integration via HACS
2. From HA: Settings -> Devices -> Add Integration
3. Select Redback Technologies Portal
4. Enter details in the dialog (device name, Redback ID, Redback Auth) and leave Redback Site as "First"
5. Your first inverter is now configured (S17*** in your list)
6. Do steps #2, #3 and #4 again but this time set Redback Site as "Second"
7. You now have both inverters registered with HA

juicejuice, 2025-08-04:
@GrantKeymer glad to hear it worked! I think you might have been able to just add the second one to the first but I couldn't test it out here. I hope the new inverter goes well.

closed 2025-08-04 by juicejuice